arendst/Tasmota #23255: Zigbee join fails for version 14.5.0

japs386, 2025-04-06:
### PROBLEM DESCRIPTION
Zigbee join fails for version 14.5.0

### REQUESTED INFORMATION
I try to join a Sonoff SNZB-02P. It will show up on the console, but the new unit will
never return a temperature or humidity.
When I downgrade to version14.4.1 a new SNZB-02P will join correctly.
All existing SNZB-02P joined under version14.4.1 will operate correctly after
an upgrade to 14.5.0

### TO REPRODUCE

### EXPECTED BEHAVIOUR

### SCREENSHOTS

### ADDITIONAL CONTEXT

s-hadinger, 2025-04-07:
Please join logs from 14.4.1 and 14.5.0 with `weblog 3` or `seriallog 3`, especially from the moment your pair the device

japs386, 2025-04-07:
The 3 new units will not fail anymore. I have used ZBLEAVE and ZBFORGET,
but I think some data are left ind the Tasmota gateway.
So I have ordered a new SNZB-02P, and will try again.

japs386, 2025-04-10:
I have tried a join with a new SNZB-02P, and is unable to
repeat the problem. Case can be closed.