arendst/Tasmota #23257: TM1638 DisplayDimmer doesn't work at all

Romik4x4, 2025-04-07:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Romik4x4, 2025-04-07:
11:31:12.202 CMD: Backlog Template; Module; GPIO 255
11:31:12.276 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
11:31:12.504 RSL: RESULT = {"Module":{"0":"Generic"}}
11:31:12.760 RSL: RESULT = {"GPIO0":{"TM1638 STB":2112},"GPIO1":{"TM1638 CLK":2048},"GPIO2":{"None":0},"GPIO3":{"TM1638 DIO":2080},"GPIO4":{"I2C SDA":640},"GPIO5":{"I2C SCL":608},"GPIO9":{"None":0},"GPIO10":{"None":0},"GPIO12":{"None":0},"GPIO13":{"None":0},"GPIO14":{"LedLink":544},"GPIO15":{"None":0},"GPIO16":{"None":0},"GPIO17":{"None":0}}

s-hadinger, 2025-04-07:
You didn't even bother to tell what hardware you are using, what version of Tasmota nor the symptoms of what you claim "doesn't work at all".

s-hadinger, 2025-04-07:
Closing as this issue is missing mandatory information

closed 2025-04-07 by s-hadinger