arendst/Tasmota #23277: Tasmota 14.5.0.3 (dfabe8c-lvgl-haspmota) build and haspmota_demo.jsonl throws out an error

jkwim, 2025-04-10:
### PROBLEM DESCRIPTION
Tasmota 14.5.0.3 (dfabe8c-lvgl-haspmota) build and haspmota_demo.jsonl throws out following error:

`00:00:01.131 HSP: invalid JSON line '{"id":1,"obj":"label","x":0,"y":30,"w%":100,"h":24,"text":"[4. Leds]","text_font":"montserrat-20"}{"id":11,"obj":"led","x":10,"y":60,"w":20,"h":20,"color":"#FF0000"}'`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [x] Device used (e.g., Sonoff Basic): ESP32-S3-WROOM-1 N16R8 with ST7735 Display
- [x] Tasmota binary firmware version number used: 14.5.0.3 (dfabe8c-lvgl-haspmota)_
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: esptool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
1. On a ESP32-S3 board with PSRAM, flash unofficial build https://github.com/tasmota/install/blob/firmware/firmware/unofficial/tasmota32s3-qio_opi.factory.bin
2. Load Haspmota demo file [haspmota_demo.jsonl](https://tasmota.github.io/docs/_media/lvgl/haspmota_demo.jsonl) and delete from Page 5 onwards
3. Add following to autoexec.be
import haspmota
haspmota.start()

### EXPECTED BEHAVIOUR
autoexec.be should get executed without any error

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
The issue was narrowed down to the length of the text field in the following statement:

{"id":1,"obj":"label","x":0,"y":30,"w%":100,"h":24,"text":"[4. Leds]","text_font":"montserrat-20"}

If the length of the text string is exactly 9 characters the error gets triggered:

00:00:01.183 HSP: invalid JSON line '{"id":1,"obj":"label","x":0,"y":30,"w%":100,"h":24,"text":"123456789","text_font":"montserrat-20"}{"id":11,"obj":"led","x":10,"y":60,"w":20,"h":20,"color":"#FF0000"}
'
**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2025-04-10:
Try again with a **very** recent build. Two hours ago, a fix looking like it could be addressing your issue was merged. https://github.com/arendst/Tasmota/pull/23276

jkwim, 2025-04-11:
Oh!
This line is exactly 98 characters:
`{"id":1,"obj":"label","x":0,"y":30,"w%":100,"h":24,"text":"[4. Leds]","text_font":"montserrat-20"}`

Will upgrade and see.

jkwim, 2025-04-11:
Confirmed working correctly with new build:
Tasmota 14.5.0.3 (cfe1cee-lvgl-haspmota)

https://github.com/tasmota/install/blob/firmware/firmware/unofficial/tasmota32s3-qio_opi.bin

closed 2025-04-11 by jkwim

s-hadinger, 2025-04-11:
It's funny. I came across this bug last evening by pure chance. It's been there for a very long time

jkwim, 2025-04-12:
Along with the help from @sfromis, I was scratching my head for several days trying to figure it out.

Then the light bulb moment came when I started to alter the length of the text field in the HASPmota example:

https://github.com/arendst/Tasmota/discussions/23244#discussioncomment-12776780

:-)