arendst/Tasmota #23289: Flaky IRSend on ESP32-C3

oogm, 2025-04-13:
### PROBLEM DESCRIPTION

#### Background

A few years back, I bought a cheap "[Nedis SmartLife WLAN IR](https://www.amazon.de/dp/B07L146TXC?ref=ppx_yo2ov_dt_b_fed_asin_title)" bridge ([amazon link](https://www.amazon.de/dp/B07L146TXC?ref=ppx_yo2ov_dt_b_fed_asin_title)) and flashed its TYWE3S with Tasmota. It's insanely reliable and accurate, to a degree where it turns of the TV within home automation for years (IRSend commands are recognized by the TV 100% of the time)

Recently, I wanted to repeat the same thing as a present for someone, but with an ESP32 ([ESP32-C3F](https://de.aliexpress.com/item/1005007029978294.html?gatewayAdapt=glo2deu)) for Matter support. Swapping chip and configuration went well, **but when sending commands via IRSend, they are only recognized by the TV (and the other IR receiver) 50% of the time. After investigating this a bit, it seems like this is because ESP32's IR send timing is less accurate.**

#### Bug
I tested with the "Volume down key". According to this [blog post](https://hawkins.id.au/posts/discrete_ir_for_hisense_tv/), this button's code is `0x00FDC23D` in the NEC protocol. According to [this definition of the NEC protocol](https://www.sbprojects.net/knowledge/ir/nec.php), we'd expect the following pulses:
```
9000, 4500, 560, 560, 560, 560, 560, 560, 560, 560,560, 560,560, 560, 560, 560, 560, 560, 560, 1690, 560, 1690, 560, 1690, 560, 1690, 560, 1690, 560, 1690, 560, 560, 560, 1690, 560, 1690, 560, 1690, 560, 560, 560, 560, 560, 560, 560, 560, 560, 1690, 560, 560, 560, 560, 560, 560, 560, 1690, 560, 1690, 560, 1690, 560, 1690, 560, 560, 560, 1690
```

We can compare these to the pulses that the Tasmota device sends by receiving them with another Tasmota-IR-bridge and decoding them with [tasmota.hadinger.fr/util](http://tasmota.hadinger.fr/util).

Doing this for both of the devices (the Nedis IR bridge with the ESP8266 and the Nedis IR bridge with the ESP32), we can easily see why where this issue comes from:

* **ESP8266**: IRSend detected by the TV 100% of the time. All pulses deviate by max 20ms from the ground truth.
* **ESP32: The TV only detects around 50% of the IRSend commands. Looking at the pulse-sequences, we can see that all non-detected commands have at least one pulse that deviates from the ground truth by 100ms (sometimes 1000ms+).** In general, both pulse value and duration seem to be less reliable.

The full measurements are in https://docs.google.com/spreadsheets/d/1XR929pzA5BJnWK5daQhsECsQqIp4_VU_jO7HdFb5j2M/view?gid=420599256#gid=420599256. I tried multiple precompiled and self-compiled versions of Tasmota (details in the sheet), there doesn't seem to be a correlation. I also bought another IR-bridge and swapped in another ESP32-C3, with the same issue, so unlikely its a hardware defect.

**TL;DR: Seems like for Tasmota on ESP32C3, IRSend's pulse timing is inaccurate.**

This is how far I got investigating this. Does anyone have an idea? Happy to help fix this / cut a PR if anyone has an idea what's going wrong here.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [x] Device used (e.g., Sonoff Basic): [Nedis SmartLife WLAN IR](https://www.amazon.de/dp/B07L146TXC?ref=ppx_yo2ov_dt_b_fed_asin_title) with [ESP32-C3F](https://de.aliexpress.com/item/1005007029978294.html?gatewayAdapt=glo2deu)
- [x] Tasmota binary firmware version number used: v11, v13, v14.5, v14.5-ir all affected
  - [x] Pre-compiled: v11, v13, v14.5
  - [x] Self-compiled: v14.5-ir (no precompiled IR binary for ESP32-C3)
- [x] Flashing tools used: originally esptool, then Tasmota-upgrade-ui
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
23:31:25.192 CMD: Backlog Template; Module; GPIO 255
23:31:25.218 RSL: RESULT = {"NAME":"Nedis IR ESP32C3","GPIO":[1,1,1,1056,1,32,1,1,1,1,1,0,0,0,0,0,0,0,288,1088,2304,2272],"FLAG":0,"BASE":1}
23:31:25.442 RSL: RESULT = {"Module":{"0":"Nedis IR ESP32C3"}}
23:31:25.644 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"1056":"IRsend1"},"GPIO4":{"0":"None"},"GPIO5":{"32":"Button1"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO18":{"288":"Led1"},"GPIO19":{"1088":"IRrecv"},"GPIO20":{"2304":"Tuya Rx"},"GPIO21":{"2272":"Tuya Tx"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
 N/A - not using rules
```
- [x] Provide the output of this command: `Status 0`:
```lua
23:32:05.515 CMD: STATUS 0
23:32:05.518 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_860730","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
23:32:05.554 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32c3.bin","RestartReason":"Software reset CPU","Uptime":"0T00:01:22","StartupUTC":"2025-04-13T22:30:43","Sleep":50,"CfgHolder":4617,"BootCount":26,"BCResetTime":"2025-04-13T21:29:08","SaveCount":53}}
23:32:05.591 RSL: STATUS2 = {"StatusFWR":{"Version":"14.5.0(release-tasmota32)","BuildDateTime":"2025-02-18T14:37:31","Core":"3_1_1","SDK":"5.3.2.250120","CpuFrequency":160,"Hardware":"ESP32-C3 v0.4","CR":"427/699"}}
23:32:05.607 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"FileLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
23:32:05.641 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":2034,"Free":845,"Heap":175,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":80,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80F1","00000815"],"Drivers":"1,2,!3,!4,5,7,!8,9,10,11,12,!14,16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,!71,!73,!75,82,!86,!87,!88,!91,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82,89"}}
23:32:05.712 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-860730-1840","IPAddress":"192.168.178.34","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"XXX","Mac":"80:65:99:86:07:30","IP6Global":"XXX","IP6Local":"XXX","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.178.1","DNSServer2":"XXX","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
23:32:05.771 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_860730","MqttUser":"DVES_USER","MqttCount":0,"MqttTLS":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
23:32:05.786 RSL: STATUS7 = {"StatusTIM":{"UTC":"2025-04-13T22:32:05Z","Local":"2025-04-13T23:32:05","StartDST":"2025-03-30T02:00:00","EndDST":"2025-10-26T03:00:00","Timezone":"+01:00","Sunrise":"06:02","Sunset":"19:38"}}
23:32:05.814 RSL: STATUS10 = {"StatusSNS":{"Time":"2025-04-13T23:32:05"}}
[... cut ...]
* Get an IR bridge/sender (most likely does not matter, since two different devices had the same issue) with an ESP32-C3
* ` IRSend {"Protocol":"NEC","Bits":32,"Data":"0xFDC23D"}` (or any other send command)
* Observe whether the IR target device or alternatively another Tasmota-IR-bridge can successfully decode the IR commands.

### EXPECTED BEHAVIOUR

1) The target device should receive IR commands 100% of the time.
2) A tasmota device should be able to receive and decode IR commands 100% of the time.

Jason2866, 2025-04-14:
It is a known issue from the used lib IRremoteESP8266 for the C3. https://github.com/crankyoldgit/IRremoteESP8266/issues/2014

Jason2866, 2025-04-22:
@dbie0 see the post before yours. It is a verified issue in the used library IRremoteESP8266.

s-hadinger, 2025-04-22:
Unfortunately the lib was never ported to support RMT which would solve this

Jason2866, 2025-04-22:
@s-hadinger There is a RMT fork https://github.com/mrninhvn/IRremoteESP/tree/dev/rmt-idf

oogm, 2025-05-03:
Hi, thanks for everyone's input on this!

I finally had the time to compile Tasmota with some of mrninhvn's changes to support RMT (at least for IRSend) from [github.com/mrninhvn/IRremoteESP/tree/dev/rmt-idf](https://github.com/mrninhvn/IRremoteESP/tree/dev/rmt-idf). I can confirm that the fork fixes the issue, and my device is indeed able to reliably execute IRsend commands.

Seems like the long term fix for this is simply waiting for mrninhvn's fork for this to (hopefully) get upstreamed into IRremoteESP8266. The last commit on mrninhvn's fork was a few weeks ago, so it seems like it is still actively being worked on.

Closing this issue, since there seems to be nothing actionable to do on the Tasmota side.

#### For reference, here's the steps I used to compile a version of Tasmota with some of mrninhvn's fixes (with the ones for IRsend, without the ones for IRrecv) in case someone wants to compile a fixed version themselves:

1. Clone and setup build environment for Tasmota
```
git clone https://github.com/arendst/Tasmota.git
cd Tasmota
# git checkout 2e251468f96cc8e253ab4b849f92d972afd9b960
python3 -m venv venv
source ./venv/bin/activate
pip install --upgrade pip && pip install --upgrade platformio
platformio upgrade &&  pio pkg update
pio run -e tasmota32c3
```

2. Add build config enabling RMT support to "platformio_tasmota_env32.ini":
```
[env:tasmota32c3-rmt]
extends                 = env:tasmota32_base
board                   = esp32c3
build_flags             = ${env:tasmota32_base.build_flags}
                          -DFIRMWARE_TASMOTA32
                          -DESP32_RMT
                          -DUSE_IR_REMOTE_FULL
lib_ignore              = ${env:tasmota32_base.lib_ignore}
                          Micro-RTSP
```

3. Swap in mrninhvn's IRremoteESP8266 fork
```
cd lib/lib_basic/IRremoteESP8266
mv IRremoteESP8266 IRremoteESP8266_OG
git clone -b dev/rmt-idf https://github.com/mrninhvn/IRremoteESP.git
mv IRremoteESP IRremoteESP8266
cd IRremoteESP8266
# git checkout a49cb0fe6496a7bf4271e48de8f25574ad40abf9
```

4. Revert RMT IRreceive changes from the fork. I didn't have time to debug it, but something in there prevents Tasmota from booting. Seems like the Arduino-based IRrecv implementation also doesn't support `pause()/resume()`, and seems like this fork started shifting to the ESP-IDF framework to address some of this. I didn't really see issues with the original IRrecv, so I didn't investigate these deeper and just reverted these parts of the fork:
    * Fully revert files "IRrecv.h", "IRrecv.cpp", and "IR_bluestar_heavy.cpp"
    * In "IRsend.h" and "IRsend.cpp" revert function "sendBluestarHeavy" 
    * In "IRremoteESP8266.h" revert all diffs containing "BluestarHeavy"

5. Compile with "pio run -e tasmota32c3-rmt"
The resulting binary indeed fixes IRSend. I've been using it for a few days now, it seems stable.

closed 2025-05-03 by oogm