arendst/Tasmota #23316: Miboxer – the ON/OFF button is not displayed

bistek22, 2025-04-18:
### PROBLEM DESCRIPTION
After updating Tasmota from version 14.2.0 to version 14.6.0, the ON/OFF button disappeared. It also doesn't work on version 14.5.0. I'm using the TUYA MCU template.
[Templates](https://templates.blakadder.com/miboxer_FUT035W.html)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [x] Device used (e.g., Sonoff Basic): Miboxer FUT035W
- [x] Tasmota binary firmware version number used: 14.6.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: esptool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### SCREENSHOTS

[image]

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2025-04-18:
Ignoring to provide the needed Infos asked in issues template no help is possible.

sfromis, 2025-04-18:
The button is still there, named "T1". Status is reflected in it changing color.