arendst/Tasmota #23324: LoRaWAN-AU915 | No Join Accept issued?

UBWH, 2025-04-21:
### PROBLEM DESCRIPTION
With AU915 devices, the LoRaWAN feature does not seem to issue correct JOIN ACCEPT messages, so the JOIN process never completes.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [x] Device used (e.g., Sonoff Basic): [LilyGo-T3S3](https://lilygo.cc/products/t3s3-v1-0) 
- [x] Tasmota binary firmware version number used: 14.6.0 (release-tasmota32)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmota Web Installer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
09:42:22.852 MQT: stat/tasmota_5390B0/RESULT = {"NAME":"LilygoT3S3-SX126x","GPIO":[1,1,1,672,1,736,704,10656,10688,1,1,1,1,1,1,1,1,1,1,1,1,1,10784,10720,1,1,0,1,1,0,1,0,1,1,1,1,1,0],"FLAG":0,"BASE":1}
09:42:23.088 MQT: stat/tasmota_5390B0/RESULT = {"Module":{"0":"LilygoT3S3-SX126x"}}
09:42:23.291 MQT: stat/tasmota_5390B0/RESULT = {"GPIO0":{"None":0},"GPIO1":{"None":0},"GPIO2":{"None":0},"GPIO3":{"SPI MISO1":672},"GPIO4":{"None":0},"GPIO5":{"SPI CLK1":736},"GPIO6":{"SPI MOSI1":704},"GPIO7":{"LoRa CS":10656},"GPIO8":{"LoRa Rst":10688},"GPIO9":{"None":0},"GPIO10":{"None":0},"GPIO11":{"None":0},"GPIO12":{"None":0},"GPIO13":{"None":0},"GPIO14":{"None":0},"GPIO15":{"None":0},"GPIO16":{"None":0},"GPIO17":{"None":0},"GPIO18":{"None":0},"GPIO19":{"None":0},"GPIO20":{"None":0},"GPIO21":{"None":0},"GPIO33":{"LoRa DIO1":10784},"GPIO34":{"LoRa Busy":10720},"GPIO35":{"None":0},"GPIO36":{"None":0},"GPIO37":{"None":0},"GPIO38":{"None":0},"GPIO39":{"None":0},"GPIO40":{"None":0},"GPIO41":{"None":0},"GPIO42":{"None":0},"GPIO43":{"None":0},"GPIO44":{"None":0},"GPIO45":{"None":0},"GPIO46":{"None":0},"GPIO47":{"None":0},"GPIO48":{"None":0}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  No Rules
```
- [x] Provide the output of this command: `Status 0`:
```lua
09:44:49.412 MQT: stat/tasmota_5390B0/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_5390B0","ButtonTopic":"0","Power":"1","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
09:44:49.415 MQT: stat/tasmota_5390B0/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32s3.bin","RestartReason":"Software reset CPU","Uptime":"0T00:47:48","StartupUTC":"2025-04-21T00:57:01","Sleep":50,"CfgHolder":4617,"BootCount":4,"BCResetTime":"2025-04-21T01:32:09","SaveCount":15}}
09:44:49.419 MQT: stat/tasmota_5390B0/STATUS2 = {"StatusFWR":{"Version":"14.6.0(release-tasmota32)","BuildDateTime":"2025-04-16T11:21:30","Core":"3_1_3","SDK":"5.3.2.250403","CpuFrequency":240,"Hardware":"ESP32-S3 v0.2","CR":"407/699"}}
09:44:49.420 MQT: stat/tasmota_5390B0/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"FileLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["UniFi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
09:44:49.429 MQT: stat/tasmota_5390B0/STATUS4 = {"StatusMEM":{"ProgramSize":1980,"Free":899,"Heap":248,"StackLowMark":3,"PsrMax":2048,"PsrFree":2016,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":80,"FlashMode":"QIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80F1","00000814"],"Drivers":"1,2,!3,!4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,73,!75,82,!86,!87,!88,!91,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82,89"}}
09:44:49.436 MQT: stat/tasmota_5390B0/STATUS5 = {"StatusNET":{"Hostname":"tasmota-5390B0-4272","IPAddress":"10.1.1.217","Gateway":"10.1.1.1","Subnetmask":"255.255.255.0","DNSServer1":"10.1.1.1","DNSServer2":"0.0.0.0","Mac":"34:B7:DA:53:90:B0","IP6Global":"fdfd:f5d8:e1fd:0:36b7:daff:fe53:90b0","IP6Local":"fe80::36b7:daff:fe53:90b0%st2","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"10.1.1.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":19.0}}
09:44:49.439 MQT: stat/tasmota_5390B0/STATUS6 = {"StatusMQT":{"MqttHost":"10.1.1.168","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5390B0","MqttUser":"ha-mqtt","MqttCount":2,"MqttTLS":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
09:44:49.442 MQT: stat/tasmota_5390B0/STATUS7 = {"StatusTIM":{"UTC":"2025-04-21T01:44:49Z","Local":"2025-04-21T09:44:49","StartDST":"2025-03-30T02:00:00","EndDST":"2025-10-26T03:00:00","Timezone":"+08:00","Sunrise":"12:47","Sunset":"02:50"}}
09:44:49.447 MQT: stat/tasmota_5390B0/STATUS10 = {"StatusSNS":{"Time":"2025-04-21T09:44:49"}}
09:44:49.451 MQT: stat/tasmota_5390B0/STATUS11 = {"StatusSTS":{"Time":"2025-04-21T09:44:49","Uptime":"0T00:47:48","UptimeSec":2868,"Heap":248,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"Berry":{"HeapUsed":3,"Objects":42},"Wifi":{"AP":1,"SSId":"UniFi","BSSId":"DC:9F:DB:1C:E0:60","Channel":11,"Mode":"HT40","RSSI":100,"Signal":-32,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
08:57:25.695 LOR: Rcvd (24718) '0000010000004140A83DAE5C4F4E4140A8924736FB4669', RSSI -121.0, SNR -15.8
08:57:25.696 LOR: JoinEUI 00010000004140A8, DevEUIh A840414E, DevEUIl 4F5CAE3D, DevNonce 4792, MIC 6946FB36
08:57:26.619 CFG: Lora saved to file
08:57:30.032 LOR: Send (29718) '20AAA4858AF512250C76178FA0FCB1A0DA', Invert 1, Time 339
08:57:31.000 LOR: Send (30719) '20AAA4858AF512250C76178FA0FCB1A0DA', Invert 1, Time 338

09:03:50.911 LOR: Rcvd (30056) '0000010000004140A83DAE5C4F4E4140A8F3F6F74F5FFF', RSSI -25.0, SNR 7.7
09:03:50.913 LOR: JoinEUI 00010000004140A8, DevEUIh A840414E, DevEUIl 4F5CAE3D, DevNonce F6F3, MIC FF5F4FF7
```

### TO REPRODUCE
Here are the details of the testing done so far.

Equipment available (all in same room): 
 * End Device: [Dragino LHT52](https://www.dragino.com/products/temperature-humidity-sensor/item/199-lht52.html)
     * AU915 model
 * LoRaWAN Gateway [Dragino LPSv2](https://www.dragino.com/products/lora-lorawan-gateway/item/228-lps8v2.html)
     * 915/920/923 MHz model
  * TWO [LilyGo-T3S3](https://lilygo.cc/products/t3s3-v1-0) 
     * Both with SX1262 915 MHz LoRa chips. 
     * Both with latest Tasmota `14.6.0 (release-tasmota32)` installed. 
	 * Both with LoRa setup as below:
```
{"LoRaConfig":
 {"Frequency":915.2,
  "Bandwidth":125.0,
  "SpreadingFactor":10,
[... 79 lines cut ...]
	
	
__BUG__

Seems in this case, Tasmota is not issuing correct JOIN ACCEPT messages.
The Tasmota LoRaWAN testing done when the feature was added to Tasmota was done on [868MHz](https://tasmota.github.io/docs/LoRa-and-LoRaWan-Bridge/#information).	

Maybe(??) the problem I am seeing is that I am testing with AU915.
[This document](https://wiki.dragino.com/xwiki/bin/view/Main/End%20Device%20Frequency%20Band/#H2.OTAAJoinBehaviour) says that Dragino products (others also??) do this:
* EU868 OTAA Join DR Logic. SF7->SF8->SF9->SF10->SF11->SF12; Each SF will make 3 attempts and then loop
* US915 and AU915 OTAA Join procedure,. End node by default will use all 72 channels (All sub-bands) to join and it will be set to use the proper sub-band after Join success

   
   

### EXPECTED BEHAVIOUR
JOIN process completes

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2025-04-21:
Tasmota cannot loop channels. In order to join commercial devices you'll have to configure them to stay on a fixed frequency as can be done with dragino hardware using it's serial interface (at least the 868 Mhz hardware I used).

UBWH, 2025-04-21:
Sure .... but that does not explain why the JOIN ACCEPT never appeared on the LPS8v2 which is kinda sniffing all frequencies.

UBWH, 2025-04-21:
Hmmmm.

In the EU LoRa Band plan, the Uplink & Downlink frequencies are the SAME: [See this](https://www.thethingsnetwork.org/docs/lorawan/frequency-plans/#eu863-870) 

But in AU, they are not: [See this](https://www.thethingsnetwork.org/docs/lorawan/frequency-plans/#au915-928)

I wonder if that is an issue in Tasmota.

Question: If a JOIN REQUEST comes into Tasmota on 915.2 MHz, what freq is the JOIN ACCEPT sent on?

arendst, 2025-04-21:
Should be 915.2 but the device will cycle 78 frequencies at different SF which makes it hard for the Lilygo single channel hardware to detect the correct one. 

I fear that different up and down channel frequencies are too much for Tasmota.

UBWH, 2025-04-21:
Think I have found the root of the problem.

The 'clue' is in this image
[image]

- The Request is on Freq 915.6 / 125 / 10
- The Accept is on Freq  924.5 / 500 / 10

That is explained [here](https://lora-alliance.org/wp-content/uploads/2020/11/RP_2-1.0.2.pdf) ... page 47

Reading that in detail, it seems with AU915 there are NO frequencies that are common to Uplink & Downlink.

- Upstream freqs are all centered on frequencies ending in even numbers (x.0, x.2, x.4, ...)
- Downstream freqs all end in odd numbers (x.1, x.3, ...)

The closest pair that (nearly) overlap I could find were:

- up:    923.8, BW 500, SF 8
- down: 923.9, BW 500, SF 8

I tried a JOIN with loraconfig with both the above option. No luck.

I think this confims that the current Tasmota (fixed freq) can never support AU915.

UBWH, 2025-04-21:
Let me ask a question ... as I am prepared to put in some effort to modify the code along these lines.

- Tasmota to support `LoRaConfigUp` and `LoRaConfigDown`
   - Inotherwords Tasmota can support different RX and TX configs
- While waiting for uplink messages: Tasmota sets the SX1262 to `LoRaConfigUp` 
- To send a downlink message:
   - Switch SX1262 to `LoRaConfigDown`
   - Send message
   - Switch SX1262 to `LoRaConfigUp`

joba-1, 2025-04-22:
Up and Down makes me dizzy - you mean: Up as receiving uplink messages - I thought: sending to uplink… so somehow I‘d prefer Rx and Tx :)

Anyways, your commands should be relatively easy copy paste programming, given LoRaConfig exists. 

Bit more difficult is the actual send which surely is async so some state machine that you would first have to find and understand has now 2 steps more.
It is not as impossible as receiving on more than one channel though.

If you get it done, I‘m sure noone will object a PR - probably with help from arendst if you tell him a bit about your hard life being upside down which now obviously culminated in strange LoRaWan rules ;)

joba-1, 2025-04-22:
I looked a bit into the code and found lora send is done synchronously, so it is not hard to change settings before and after sending. This poc hack might already make it work for au devices: #23330

UBWH, 2025-04-22:
Actively working on a fix as we speak for AU915 (without breaking existing behaviour for EU)

Give me a few days.

UBWH, 2025-04-25:
@arendst 

Theo ... I have made progress but current problem is that after each power cycle - I only receive ONE uplink message. 

OBJECTIVES - SUPPORT AU915

- Add concept of LoRa Bands
  - Default = EU868  (Band 0)
  - New     = AU915  (Band 1)
  - SUCCESS. This is how Tasmota knows what valid Downlink frequencies to use.

```
"LoRaConfig":{
  "Frequency":915.2,
  "Bandwidth":125.0,
  "SpreadingFactor":10,
  "CodingRate4":5,
  "SyncWord":52,
  "OutputPower":10,
  "PreambleLength":8,
  "CurrentLimit":60.0,
  "ImplicitHeader":0,
  "CrcBytes":2,
  "Band":"AU915"     <---------New
}
```
- Send  JOIN ACCEPT on different Freq that the JOIN REQUEST received on
  - SUCCESS
  - Device stops sending JOIN REQUESTS and starts sending DATA UPLINKS
- Use JOIN ACCEPT `CFlist` to instruct device to only use one Uplink Freq
  - SUCCESS

[image]

I have added a lot of DBG messages to see what is going on

Uplink message received.
```
08:45:46.113 S6X: Flag (644163) irq 0002
08:45:46.113 DBGROB: LoraSx126xReceive() Re-setting RXflag
08:45:46.115 DBGROB: Rcvd packets 15 bytes
08:45:46.115 LOR: Rcvd (644162) '40709B5300C141000D02C681FAB488', RSSI -21.0, SNR 7.5
08:45:46.115 ROBLOR: Calling(LoRaWanInput())
08:45:46.115 DBGROB: Got Data Uplink. DevAddr = 5479280
08:45:46.118 LOR: DevAddr 00539B70, FCtrl C1, FOptsLen 1, FCnt 65, FOpts 0D, FPort 2, Payload C6, Decrypted 00, MIC 88B4FA81
08:45:46.118 LOR: Missed frames 64
08:45:46.118 DBGROB:LoraWanSendMacResponse(node = 0)
08:45:46.120 DBGROB:LoraWanSendResponse() Delay = 1000 LoraBand=1
08:45:46.123 MQT: tele/tasmota_539B70/SENSOR = {"LwReceived":{"0xAE3D":{"Node":1,"Device":"0xAE3D","RSSI":-21.0,"SNR":7.5,"DevEUIh":"A840414E","DevEUIl":"4F5CAE3D","FPort":2,"Payload":[0]}}}
08:45:46.971 CFG: Lora saved to file
```
Then Tasmota sends the response to the `DeviceTimeAns` MAC command requested in the Uplink message.
```
08:45:47.113 DBGROB:LoraWanTickerSend() send_buffer_step=2
08:45:47.113 DBGROB: LoraSend() Changing to TX profile. Current Freq:915.2  
08:45:47.113 DBGROB:LoraDefaults(AU915) uBand=1 bRX=0 channel=0
08:45:47.114 DBGROB:LoraDefaults(AU915) TX: uRxChannel=0, new TX channel:0
08:45:47.115 DBGROB:LoraSx126xConfig() f=923.3
08:45:47.418 LOR: Send (645170) '60709B53002600000DA99D355500A1B6686A', Invert 1, Time 298
08:45:47.418 DBGROB: LoraSx126xAvailable(1) Re-setting RXflag
08:45:47.418 DBGROB: LoraSend() Changing to RX profile Current Freq:923.3  
08:45:47.418 DBGROB:LoraSx126xConfig() f=915.2

08:45:48.113 DBGROB:LoraWanTickerSend() send_buffer_step=1
08:45:48.113 DBGROB: LoraSend() Changing to TX profile. Current Freq:915.2  
08:45:48.114 DBGROB:LoraDefaults(AU915) uBand=1 bRX=0 channel=0
08:45:48.115 DBGROB:LoraDefaults(AU915) TX: uRxChannel=0, new TX channel:0
08:45:48.115 DBGROB:LoraSx126xConfig() f=923.3
08:45:48.424 DBGROB: LoraSx126xAvailable(1) Re-setting RXflag
08:45:48.424 LOR: Send (646176) '60709B53002600000DA99D355500A1B6686A', Invert 1, Time 298
08:45:48.424 DBGROB: LoraSend() Changing to RX profile Current Freq:923.3  
08:45:48.425 DBGROB:LoraSx126xConfig() f=915.2
```

.... after this .... no more uplink messages 'seen' by Tasmota. I have checked Lora->send_flag & Lora->received_flag as below

```
10:22:38.397 ROBDBG: SendFlag:0, ReceivedFlag:0
```

Any clues on what what would prevent new incoming messages been seen?

UBWH, 2025-04-27:
__SUCCESS!__

In summary:
- Added support for AU915
  - this necessitated supporting Tasmota to transmit downlink responses on different Freq/BW/SF than the received Uplink

__Logic (AU915)__
- Default: SX126x set to RX Freq/BW/SF values
- If sending a Downlink back to the Device:
   - Configure SX126x to TX Freq/BW/SF values
   - Send Downlink
   - Configure SX126x to RX Freq/BW/SF values

__Roadblock__
With the above logic, would only receive a few (1 to 5?) Uplinks - then nothing until next Reboot

__The Fix__
 Temporary code - just for testing with my SX1262 
```
void LoraWanTickerSend(void) {
  Lora->send_buffer_step--;
  if (1 == Lora->send_buffer_step) {
    Lora->rx = true;                                // Always send during RX1
    Lora->receive_time = 0;                         // Reset receive timer
    LoraWan_Send.once_ms(TAS_LORAWAN_RECEIVE_DELAY2, LoraWanTickerSend);  // Retry after 1000 ms
  }

  if (Lora->rx) {                                   // If received in RX1 do not resend in RX2
    LoraSend(Lora->send_buffer, Lora->send_buffer_len, true);
  } else {
    Lora->send_buffer_step = 0; //Nothing more to send
  }

  //// TO DO: Remove hardcoded SX126X function (in case SX127X)

  if(0 == Lora->send_buffer_step){
    LoraSx126xInit();                             <-------------- This was the Fix. Basically kick start the SX126x
  }
}
```
@arendst: Do you have any concerns issues with this technique?

arendst, 2025-04-27:
> Do you have any concerns issues with this technique?

Not if we can make it specifig for AU915. I suggest you enetr a PR and I will see how to integrate it that with the current 868 version.

UBWH, 2025-05-07:
Resolved with https://github.com/arendst/Tasmota/commit/b5ae6bc035f92baa035c29692904e6572dd0518d