arendst/Tasmota #23340: Shelly Plus 1PM Mini Serial Flash Problem After "MgOS32 to Tasmota32" Failed

jvincze84, 2025-04-25:
### PROBLEM DESCRIPTION

I have a Shelly Plus 1PM mini module.  (SNSW-001P8EU)

I have read this [https://templates.blakadder.com/shelly_plus_1pm_mini.html](https://templates.blakadder.com/shelly_plus_1pm_mini.html)

First I have tried the OTA flash method, as the link above suggested.  

I could access the web interface on http://192.168.4.1, but after I tried to set up wifi connection, I got Successfully Connected message, but the IP address was 0.0.0.0. 

I checked my router DHCP leases, but I faild to find the device. I restarted the device. 

After this point I cannot access my device.

I decided to flash with serial connection. 

Details of the device:

```
sudo python3 esptool.py --port /dev/ttyUSB0 flash_id
esptool.py v4.7-dev
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE, Embedded Flash 4MB (XMC)
Crystal is 40MHz
MAC: 54:32:04:ab:9d:b4
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

I tried the following binaries:

* tasmota32.bin
* tasmota32c3.bin
* tasmota32.factory.bin
* tasmota32c3.factory.bin

None of worked. 

I tried with the webinstaller (https://tasmota.github.io/install/) and with epstool.py.
Both of them give me success message, but after the flashing I can't find the Wifi AP. 
(With the web installer I tried not to specify custom image, as well).

Every help would be appreciated. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [ ] Device used (e.g., Sonoff Basic): _____
- [X] Tasmota binary firmware version number used: tasmota32.factory.bin
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: WebFlasher, esptool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:

  I cannot access the devicve

- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2025-04-25:
From your list, the only one with any chance of working would be `tasmota32c3.factory.bin`, which is also what the *recommended* Tasmota Web Installer should pick automatically.

One clear risk factor is that your undisclosed USB/serial adapter may not be able to provide the power for Wifi (this often happens with ESP32 chips), where the recommended step is to remove this wiring, and instead get power from mains.

jvincze84, 2025-04-25:
Hi @sfromis,

Thanks for the reply.
I think find something.

When I put RST to GND and release I get this message:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0x5 (DOWNLOAD(USB/UART0/1))
waiting for download
```

The GPIO9 is not connected to GND.

Do you have any idea how to get out from this?

Thanks.

jvincze84, 2025-04-25:
> Did you try what I suggested, disconnecting the wiring for flashing, and instead powering via mains?
> 
> AFAIK, that "waiting for download" means ready to go for flashing, but if you already did that (with the correct image / method), there should be no need for doing it again.

Yea, I have tried disconnecting the wiring and powering via mains, but the wifi ap did not show up.