arendst/Tasmota #23350: Bistable relays are triggered when other relays are activated / deactivated

sorivanu, 2025-04-28:
DO NOT DELETE ANY TEXT from this template! Otherwise the issue will be auto-closed.
-->

### PROBLEM DESCRIPTION
Bistable relays are triggered (pulse to deactivate the relays) when other relays buttons are pressed or console command is set.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [ ] Device used (e.g., Sonoff Basic): ESP32 NodeMCU
- [ ] Tasmota binary firmware version number used: 14.6.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: [From ](https://tasmota.github.io/install/)
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

[Config_tasmota_F57A10_6672_14.6.0.dmp](https://github.com/user-attachments/files/19936067/Config_tasmota_F57A10_6672_14.6.0.dmp)

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here: No Rules

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

[status0.txt](https://github.com/user-attachments/files/19936059/status0.txt)

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Assign GPIO22 to RELAY_B 1 
Assign GPIO23 to RELAY_B 2 
Assign GPIO18 to RELAY_B 3
Assign GPIO19 to RELAY_B 4
Assign GPIO14 to RELAY 5
Assign GPIO15 to RELAY_i 6
Assign GPIO12 to RELAY 7

Press web button to activate the any relay. 

### EXPECTED BEHAVIOUR
When any button is pressed the bistable relay are not triggered

### SCREENSHOTS

[image]

[image]

[image]

[image]

### ADDITIONAL CONTEXT
It seems that if the RELAY_b is used, when any command is send / any button is pressed to activate / deactivate other relays (bistable or not) all bistable relays trigger an deactivation pulse.  

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2025-04-29:
Thx for the observation.

As this is a theoretical exercise I hope in practice where ONLY bistable OR non-bistable relays are used there is no issue.

sorivanu, 2025-05-05:
Hi, I was hoping to use both bistable and normal relays but if this is not possible I will try a workaround. 
Thank you!