arendst/Tasmota #23383: Sonoff POWR316 continually reboots once flashed with Tasmota with error rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

paulhollands, 2025-05-06:
### PROBLEM DESCRIPTION
Flashed a new Sonoff POWR316 with latest version of Tasmota32.bin via ESPHome-Flasher and it successfully flashes, but then will not boot and a Tasmota names Wifi isn't ever available to enable configuration.

Have tried on two different POWR316 new devices and both have the same issue 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [ ] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: _14.6.0____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: _ESPHOME-FLASHER____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
[16:24:41]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:24:41]configsip: 0, SPIWP:0xee
[16:24:41]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:24:41]mode:DIO, clock div:2
[16:24:41]load:0x3fff0018,len:4
[16:24:41]load:0x3fff001c,len:1044
[16:24:41]load:0x40078000,len:8896
[16:24:41]load:0x40080400,len:5828
[16:24:41]entry 0x400806ac
[16:24:41]ets Jul 29 2019 12:21:46
[16:24:41]
[16:24:41]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:24:41]configsip: 0, SPIWP:0xee
[16:24:41]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:24:41]mode:DIO, clock div:2
[16:24:41]load:0x3fff0018,len:4
[16:24:41]load:0x3fff001c,len:1044
[16:24:41]load:0x40078000,len:8896
[16:24:41]load:0x40080400,len:5828
[16:24:41]entry 0x400806ac
[16:24:42]ets Jul 29 2019 12:21:46
[16:24:42]
[16:24:42]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:24:42]configsip: 0, SPIWP:0xee
[16:24:42]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:24:42]mode:DIO, clock div:2
[16:24:42]load:0x3fff0018,len:4
[16:24:42]load:0x3fff001c,len:1044
[16:24:42]load:0x40078000,len:8896
[16:24:42]load:0x40080400,len:5828
[16:24:42]entry 0x400806ac

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

paulhollands, 2025-05-06:
Sorry, to further add; the error is seen via ESPHome console logs and the device is still powered by the 3.3v USB to UART. I've applied mains power to the device (of course without the computer connected the same time) and this is when I never get a Tasmota wifi presented to further configure. Unfortunately no logs at this point since I can never connect to it.

arendst, 2025-05-07:
Search this site.

Hints:
1) Never use ESPHOME-Flasher for Tasmota. Use Tasmota Web installer which will select the correct binary
2) Binary `tasmota32.bin` should not be initialy flashed.