arendst/Tasmota #23388: LoRaWAN AU915: Can set illegal TX frequency

UBWH, 2025-05-07:
### PROBLEM DESCRIPTION
Command `Loraconfig 41` sets transmission on illegal freq

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [X] Device used (e.g., Sonoff Basic): __LilyGoT3S3___
- [X] Tasmota binary firmware version number used: _"Version":"14.6.0.1(75b109c-tasmota32)"
  - [X] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _n/a____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
Not needed

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
No Rules

```
- [ ] Provide the output of this command: `Status 0`:
```lua
 Not needed

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
Not needed

```

### TO REPRODUCE
```
<Reset Settings>
<Load DIO Template>
<Delete hidden file .drvset000>

01:08:16.510 CMD: loraconfig
01:08:16.513 RSL: RESULT = {"LoRaConfig":{"Region":"EU868","Frequency":868.0,"Bandwidth":125.0,"SpreadingFactor":9,"CodingRate4":7,"SyncWord":18,"OutputPower":10,"PreambleLength":8,"CurrentLimit":60.0,"ImplicitHeader":0,"CrcBytes":2}}

01:46:40.401 CMD: loraconfig 41
01:46:40.409 RSL: RESULT = {"LoRaConfig":{"Region":"AU915","Frequency":[915.0,923.3,923.3],"Bandwidth":[125.0,500.0,500.0],"SpreadingFactor":[9,9,12],"CodingRate4":7,"SyncWord":18,"OutputPower":10,"PreambleLength":8,"CurrentLimit":60.0,"ImplicitHeader":0,"CrcBytes":2}}
```
-  Lora mode (e.g. `LoraSend Hello World`): the Transmit center freq is set to 915.0, with a 0.125 MHz bandwidth = 914.9375 to 915.0625 MHz. That is illegal in AU. 

- In LoraWAN mode: not an issue as 923.3 & 923.3 used

Reference: https://www.legislation.gov.au/F2015L01438/latest/text
    Table 1, Column 2
		   
	Column 2: Permitted operating frequency band (MHz) (lower limit exclusive, upper limit inclusive)
				            
				20 All transmitters  915–928  3 mW
				
Note the words: "lower limit exclusive". Even 915.0 is not permitted
		

### EXPECTED BEHAVIOUR
Not set to illegal TX freq

Loraconfig 41 should set Frequency":[915.2,923.3,923.3],Bandwidth":[125.0,500.0,500.0] = 915.1375 to 915.2625 MHz = Legal on AU.
					  
In other words .. Loraconfig 41 and Loraconfig 42 should both do the same thing

My original command (Loraconfig AU915) did that for this reason.	:-)

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2025-05-07:
So I think that's just a matter of setting the default LoRa frequency define TAS_LORA_AU915_FREQUENCY.

How about changing this from 915.0 to 915.2?

UBWH, 2025-05-07:
Yep - that should work.

But it still means that the behaviour of
`loraconfig 41` and `loraconfig 42` are same

Still prefer my  `loraconfig AU915 n` where n = channel (defaults to 0)

As it is now, a user wanting to move to a different permitted AU LoRaWAN channel, is expected to read [this](https://lora-alliance.org/wp-content/uploads/2020/11/lorawan_regional_parameters_v1.0.3reva_0.pdf)

If LoRaWAN user does this `loraconfig {"Frequency":915.3}`: 915.3 is not allowed for LoRaWAN in AU... so Tasmota will be listening on a freq never used by AU915 devices.  (Of course: OK still for `LoRaSend Hello World`)

arendst, 2025-05-07:
So for AU915 LoRaWan it makes sense to disable user input for frequency but add a JSON option to select a channel between 0 and 71 like 
```
LoRaConfig {"Channel":10}
```
which in turn will set the frequency to an allowed value.

For Lora the user then can still change the frequency to any value he/she wishes.

arendst, 2025-05-07:
After some thought I'll add the channel as an option to `LoraConfig 41,10` now. This will set the frequency accordingly.

UBWH, 2025-05-07:
`LoraConfig 41,10`

That will work

`LoraConfig 41` should default to chanenel 0
`LoraConfig 41,nn` should be limited to 63 max

`if (uChannel > 63) uChannel = 63;   `

Here is a the code I commited on this:

```
void LoraRadioInfo(uint8_t mode, void* pInfo, uint8_t uChannel=0) {
  LoRaRadioInfo_t* pResult = (LoRaRadioInfo_t*) pInfo;
 
  if ( TAS_LORA_REGION_AU915 == Lora->settings.region ) {
    //////////////// AU915 ////////////////////
    /*    ref: https://lora-alliance.org/wp-content/uploads/2020/11/RP_2-1.0.2.pdf page 47
     DR0  LoRa: SF12 / 125 kHz 
     DR1  LoRa: SF11 / 125 kHz 
     DR2  LoRa: SF10 / 125 kHz   <-- JOIN REQUEST 
     DR3  LoRa: SF9  / 125 kHz 
     DR4  LoRa: SF8  / 125 kHz 
     DR5  LoRa: SF7  / 125 kHz 
     DR6  LoRa: SF8  / 500 kHz        Same as DR12
     DR7  LR-FHSS CR1/3: 1.523 MHz OCW 162
     DR8  LoRa: SF12 / 500 kHz
     DR9  LoRa: SF11 / 500 kHz 
     DR10 LoRa: SF10 / 500 kHz 
     DR11 LoRa: SF9  / 500 kHz 
     DR12 LoRa: SF8  / 500 kHz        Same as DR6
     DR13 LoRa: SF7  / 500 kHz 

     UPLINK (RX) CHANNELS
     There are 72 channels
       0-63: DR0 to 5. Starting 915.2, incrementing by 0.2 Mhz to 927.8  <-- JOIN REQUEST
      64-71: DR6     . Starting 915.9, incrementing by 1.6 MHz to 927.1
      NOTE: Testing with two Dragino end devices shows they do not play nice with channels 64-71
          1) LHT52 will JOIN OK on Ch64, but never sends any sensor messages on same channel
          2) LHT65 will not even broadcast JOIN requests on any of the channels (as required by RP002)
          For this reason, channels above 63 are not permitted.

     DOWNLINK (TX) CHANNELS
     There are 8 channels
       0-7: DR8 to 13. Starting 923.3, incrementing by 0.6 MHz to 927.5

     After an uplink:  Downlink (TX) link subchannel = Uplink (RX) Channel Number modulo 8   
     e.g. --Uplink (RX)--     --Downlink (TX)--
          Freq    Channel     Channel Frequency
          915.2   0           0       923.3
          927.8   63          7       927.1

     After an uplink:  
     Downlink DR for RX1 must follow this table
     Uplink  Downlink
     DR0     DR8
     DR1     DR8
     DR2     DR10   <----- channels 1-62
     DR3     DR11
     DR4     DR12
     DR5     DR13
     DR6     DR13  <------ channels 63-71

     Downlink DR for RX2 must be DR8

     Downlink
      Reference: https://lora-alliance.org/wp-content/uploads/2020/11/lorawan_regional_parameters_v1.0.3reva_0.pdf)
      Assume this is in response to an uplink RX, so we already know RX freq & bw & sf
      TX channel depends on RX freq
      DR for RX1 depends on incoming DR (and RX1DROffset)
      DR for RX2 is fixed ar DR8 
      
      Tasmota does not support different RX1 & RX2 DRs (yet), so just use DR8 and rely on RX2 arriving at end device OK.
    */
    uint8_t UplinkChannelBand = LoraChannel()%8; //0..7

    if (TAS_LORA_RADIO_UPLINK == mode) {
    //if (uChannel > 71) uChannel = 71;   See note above
      if (uChannel > 63) uChannel = 63;   
      
      if (uChannel < 64){
          (*pResult).frequency        = TAS_LORA_AU915_FREQUENCY_UP1 + (uChannel * 0.2);
          (*pResult).bandwidth        = TAS_LORA_AU915_BANDWIDTH_UP1;               //DR2
          (*pResult).spreading_factor = TAS_LORA_AU915_SPREADING_FACTOR_UP1;        //DR2
      } else {
          (*pResult).frequency        = TAS_LORA_AU915_FREQUENCY_UP2 + ((uChannel-64) * 1.6);
          (*pResult).bandwidth        = TAS_LORA_AU915_BANDWIDTH_UP2;                //DR6
          (*pResult).spreading_factor = TAS_LORA_AU915_SPREADING_FACTOR_UP2;         //DR6
      }
```

eyestorefivemstore, 2025-05-10:
Interesting solution to this problem! I've been working on similar challenges recently.

eyestoreoye, 2025-05-10:
This is a well-structured codebase. I appreciate the clean separation of concerns.

closed 2025-05-10 by UBWH