arendst/Tasmota #23416: Setting the module to Dimmer renders the button and ledlink non-functional

llambkin, 2025-05-14:
Hey Tasmota team,

I have this touch activated switch:
https://templates.blakadder.com/deta_6911HA.html

It works just fine as a simple switch, when I set up the devgroups and devgroupshare. But I wanted to activate the ability to dim the light by holding the button down, then brighten the light by holding it again, single press for on/off.

I have set all the correct options, but I have found I can't use the DIMMER function without setting the dimmer module as active. But when I do set this, it doesn't matter what I set the GPIO pins to, the button becomes non- functional, and the WiFi status light becomes not lit.

I can operate the devgroup light and brightness works from the web console, but the hardware button does not function. I cannot get the console to output any data showing I have pressed the button either.

Expectation:
Set dimmer module in Configuration/module
Configure template to match GPIO pins of template in blackadder
Configure devgroups/sharing etc., configure MQTT, setoptions for switchmode etc.
Be able to use the button to dim and brighten the devices in the devgroup, and WiFi status light work as per configured template.

Please let me know if you need any more information

sfromis, 2025-05-14:
That template looks like a very basic on/off switch, not in any way a dimmer. Without a local dimmer, it cannot **share** the dimming with other devices, but configuring a virtual PWM on an unused gpio could have a chance of working out.

However, if by "using the dimmer function" you **instead** mean using the predefined module config "PWM Dimmer", that config will totally remap gpios to fit very specific hardware, Martin Jerry/acenx/Tessan/NTONPOWER SD0x PWM dimmer switches, which is not what you have. Won't work.

If you want to experiment with the PWM dimming feature independently of that specific hardware, don't use "Configuration/Module" at all, but do it all in "Configuration/Template", where you can select it in "Based on", along with your gpios.

llambkin, 2025-05-14:
Thankyou for your reply!

Yes I know the switch plate I have is a basic on/off relay switch. I currently have no load plugged into the relay, and use the button to turn a light on and off with device groups.

I have been trying to get the button to act as a dimmer using rules, but when I add a rule, it states "unknown command" when I attach the rule.

I will give the "based on" approach a try, using switchmode 11 and some rules, and see how I go.

Sorry about posting this in the "Issue", I will post in "discussions" in future. In my defence, I thought it was an error, because the documentation states for dimmers I had to use the dimmer module, and every time I activated Module type PWM Dimmer, my switch plate would just stop responding to buttons. It would be nice to know what the PWM dimmer Module actually changes in the configuration on that wiki page so people would be informed about what they need to use.

[image]

sfromis, 2025-05-14:
Don't mix up SwitchMode 11 and PWM Dimmer support, as it is not part of the same, instead incompatible ways of aiming for similar purposes.