arendst/Tasmota #23458: Berry MQTT subscription gets lost after few hours.

Mape24, 2025-05-22:
Hi,
I got MQTT subscription running smoothly in Berry for the first few hours. After ~12h the engine simply stops calling the MQTT callback function. Berry continues to work without MQTT data. It seems to be a bug. Debugging is time consuming. I will add more information over the next weeks.

```berry
import json
import mqtt
import string
var init=0                       # wir müssen MQTT verzögert initialisieren
var Batt_Volt=0

# ----- MQTT callback function:
#def PowMr_MQTT_data(topic, idx, payload, bindata)
	if string.find(topic,"PowMr") > 0
		Batt_Volt    = json.load(payload)['Batt']['Batt_Volt']
		print("+++ MQTT received from", topic, payload, Batt_Volt)
	end
end

def f_10sec() 
# ----- Initialisierung:
	if init == 0                          # verzögert starten!
		init = 1
		mqtt.subscribe("stat/PowMr/RESULT", PowMr_MQTT_data)         # MQTT muß man verzögert starten!
	end
end

tasmota.add_cron("*/10 * * * * *", f_10sec, "every_10_s")
``` 

HW: ESP32C3
FW: Tasmota 14.6.0.1 self compiled

sfromis, 2025-05-22:
Actually, you do not need to delay the MQTT subscribe, since a long time ago the `mqtt` module will accept a `mqtt.subscribe` any time, and make the broker call when relevant, including after reconnecting. I'd recommend to not have that `cron` timer, and just do the `mqtt.subscribe` right away.

And if you actually needed the delay, a more robust approach would be new new feature of `tasmota.when_network_up`, to provide a function to be executed when networking is available. But not needed for MQTT subscriptions.

Back to the topic, one step in debugging could be to inspect the list `mqtt.topics` (like just printing it) which should contain one listener entry per subscription. It could also be relevant to check the uptime, like if there was a restart for some reason.

Mape24, 2025-05-24:
This suggested version of code (see below) does not work. So I went back to the code version above. 
Meanwhile I use FW Tasmota 14.6.0.2 self compiled. MQTT still gets lost after 12-20h. 
Recurrent entering 
`<mqtt.subscribe("stat/PowMr/RESULT", cb_MQTT_data)`
does not fix the issue. I will try `mqtt.topics` next time it happens. 

none working code:
```berry
# rename to autoexec.be 

import json
import mqtt
import string

var Batt_Volt=0

# ----- MQTT callback function for battery data:
def cb_MQTT_data(topic, idx, payload, bindata)
	if string.find(topic,"PowMr") > 0  && string.find(payload,"Batt") > 0
		Batt_Volt    = json.load(payload)['Batt']['Batt_Volt']
		print("+++ MQTT from", topic, payload, Batt_Volt)
	end
end

mqtt.subscribe("stat/PowMr/RESULT", cb_MQTT_data)
```

sfromis, 2025-05-24:
Doing a test with your "none working code" (with different strings to match my setup), I saw the subscription work as expected, without that fluff about wrapping in a `cron` timer.

Right after boot, I'm getting `MQT: Subscribe to xxxx` (but obviously no live subscription yet), and after connecting:
```
BRY: mqtt subscribe all registered topics
MQT: Subscribe to xxxx
```

Detail: If you want to check if a string contains some substring, it would be more reliable with `> -1` as the result will be 0 if the substring is contained in the start.