arendst/Tasmota #23459: ESP32S3 serial not working in Berry

LesNewell, 2025-05-22:
### PROBLEM DESCRIPTION
Attempting to open a serial port appears to have no effect in Berry. For instance if I expect TX to be on 20 and RX to be on 19 I would use this code:
`ser = Serial(19, 20, 9600)`
Attempting to send data through the serial port has no effect:
`ser.write(10)`
Looking at pin 20 with an oscilloscope I see no data. To confirm I am looking at the correct pin I configured it as PWM and observed the expected wave form.

I intially came across this problem while tring to run Zigbee using a CC2562P. Again I saw no data being sent out of the TX pin.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [ x] Device used (e.g., Sonoff Basic): Crowpanel Advance 7
- [ ] Tasmota binary firmware version number used: Self compiled latest Git and pre-compiled release ESP32S3
  - [ x] Pre-compiled
  - [ x] Self-compiled
- [ x] Flashing tools used: PlatformIO on VSCode
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
16:35:10.896 RSL: RESULT = {"NAME":"ESP32S3","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,0,0,0,0,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":1}
16:35:11.116 RSL: RESULT = {"Module":{"1":"ESP32S3"}}
16:35:11.371 RSL: RESULT = {"GPIO0":{"Option A3":6210},"GPIO1":{"None":0},"GPIO2":{"Zigbee Rst":5792},"GPIO3":{"None":0},"GPIO4":{"None":0},"GPIO5":{"None":0},"GPIO6":{"None":0},"GPIO7":{"None":0},"GPIO8":{"Zigbee Rst":5793},"GPIO9":{"None":0},"GPIO10":{"None":0},"GPIO11":{"None":0},"GPIO12":{"None":0},"GPIO13":{"None":0},"GPIO14":{"None":0},"GPIO15":{"I2C SDA1":640},"GPIO16":{"I2C SCL1":608},"GPIO17":{"None":0},"GPIO18":{"None":0},"GPIO19":{"None":0},"GPIO20":{"None":0},"GPIO21":{"None":0},"GPIO33":{"None":0},"GPIO34":{"None":0},"GPIO35":{"None":0},"GPIO36":{"None":0},"GPIO37":{"None":0},"GPIO38":{"None":0},"GPIO39":{"None":0},"GPIO40":{"None":0},"GPIO41":{"None":0},"GPIO42":{"None":0},"GPIO43":{"None":0},"GPIO44":{"None":0},"GPIO45":{"None":0},"GPIO46":{"None":0},"GPIO47":{"None":0},"GPIO48":{"None":0}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
16:36:39.187 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_140420","ButtonTopic":"0","Power":"1","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
16:36:39.193 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32s3.bin","RestartReason":"Software reset CPU","Uptime":"0T00:26:18","StartupUTC":"2025-05-22T15:10:21","Sleep":50,"CfgHolder":4617,"BootCount":48,"BCResetTime":"2025-05-20T19:44:10","SaveCount":160}}
16:36:39.197 RSL: STATUS2 = {"StatusFWR":{"Version":"14.6.0(release-tasmota32)","BuildDateTime":"2025-04-16T11:21:30","Core":"3_1_3","SDK":"5.3.2.250403","CpuFrequency":240,"Hardware":"ESP32-S3 v0.2","CR":"425/699"}}
16:36:39.202 RSL: STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"MqttLog":0,"FileLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TP-LINK OFFICE","House2"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
16:36:39.214 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1980,"Free":1923,"Heap":218,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":16384,"FlashSize":16384,"FlashChipId":"1840C8","FlashFrequency":80,"FlashMode":"QIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80F1","00000814"],"Drivers":"1,2,!3,!4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,!27,29,34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,!73,!75,82,!86,!87,!88,!91,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82,89"}}
16:36:39.222 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-140420-1056","IPAddress":"192.168.2.227","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"98:88:E0:14:04:20","IP6Global":"fdfc:90ec:5f2c:0:9a88:e0ff:fe14:420","IP6Local":"fe80::9a88:e0ff:fe14:420%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":19.0}}
16:36:39.229 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_140420","MqttUser":"DVES_USER","MqttCount":0,"MqttTLS":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
16:36:39.233 RSL: STATUS7 = {"StatusTIM":{"UTC":"2025-05-22T15:36:39Z","Local":"2025-05-22T16:36:39","StartDST":"2025-03-30T02:00:00","EndDST":"2025-10-26T03:00:00","Timezone":"+01:00","Sunrise":"04:59","Sunset":"20:33"}}
16:36:39.241 RSL: STATUS10 = {"StatusSNS":{"Time":"2025-05-22T16:36:39"}}
16:36:39.249 RSL: STATUS11 = {"StatusSTS":{"Time":"2025-05-22T16:36:39","Uptime":"0T00:26:18","UptimeSec":1578,"Heap":216,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":27,"MqttCount":0,"Berry":{"HeapUsed":5,"Objects":61},"Wifi":{"AP":1,"SSId":"TP-LINK OFFICE","BSSId":"00:25:86:BC:D8:EA","Channel":6,"Mode":"11g","RSSI":100,"Signal":-42,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Monitor pin 20 with an oscilloscope or serial port.
In Berry enter this code:
<ser = Serial(19, 20, 9600)
ser.write(10)>
No data is observed on pin 20

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Expected to see data coming out of pin 20

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

LesNewell, 2025-05-22:
Thank you for looking at this. I just tested pins 4,5 and it worked. When I originally tested these pins I misread the board schematic so I was looking at the wrong pins. However I am unable to get this code to work with pins 19,20. On this board the pins are not used for USB. Is there something I need to change in my override when I build Tasmota?

sfromis, 2025-05-22:
I'd guess that gpio 19/20 are still assigned for USB CDC usage, even with nothing connected. If you configure these pins as "Input" (instead of leaving them as "None" / "User"), this will be overridden, making them available for functions outside of core Tasmota, like Berry.

LesNewell, 2025-05-22:
I haven't tried assigning them as input but assigning them as Zigbee tx/rx did not work. That's what initially led me down this rabbit hole. I was trying to get Zigbee working.

sfromis, 2025-05-22:
Well, try my suggestion then. Only assign for Zigbee if you actually want to use Z2T to have Tasmota manage a coordinator function, where it would not be relevant to use the pins in Berry.

If it was only for verification, with you instead trying to make Zigbee work on the pins, that's a different ball game, as next step.

LesNewell, 2025-05-22:
It was for verification as I was trying to get a Zigbee module running. I needed Berry to flash the module firmware. I do now have it up and running on pins 4,5.
This board has a connector marked 'uart1' that is wired to use pins 19/20 so it would be nice to get those pins working with serial.

sfromis, 2025-05-22:
I'd call it a bad board design to expect gpio 19/20 to be "good" UART pins, there is no reason to make that assumption, certainly not something "preferable" for ESP32-S3, au contraire.

Of course, part of the story is that the large majority of ESP32-S3 boards connecting something to gpio 19/20 will be taking advantage of the MCU having built-in USB for those two pins. And therefore Tasmota default setup is to try to get USB CDC up at boot, using the pins.

Not sure how this might interfere with using them for Zigbee. In principle, configuring the pins for this "should" sidestep the USB CDC part, but such outer edge cases may not have been fully tested.

sfromis, 2025-05-22:
While Tasmota will look for USB CDC, if not detected first message after boot should be:
> `CMD: Fall back to serial port, no SOF packet detected on USB port`

When I suggested configuring as "Input" for Berry serial to use gpio 19/20 successfully, I did verify that this worked (and only when making this config), but I still got the above message indicating that it did look for having CDC available on the other end.

Having an ESP32-S3 on a board with two USB-C connectors, one connected directly to gpio 19/20, and the other via a USB/Serial adapter chip, I'm seeing UART serial console work, at least when when powering up (without anything connected to the native USB port).

Most ESP32-S3 builds includes `-DARDUINO_USB_MODE=1 -DUSE_USB_CDC_CONSOLE`, but there are also board definitions without these options.

LesNewell, 2025-05-23:
This is running a 7" display.