arendst/Tasmota #23526: PulseTime - Remaining time resets to Set time after restart

Arun-R-S, 2025-06-09:
Hi,

In tasmota I am using the pulseTime feature to turn off a specific switch after a particular second but the remaining time is reset back to Set time after a restart.

For example if I set the PulseTime1 as 180 seconds and turn on the Power1, the switch 1 is turn on and the Remaining time in the PulseTime1 start decreasing. In a case the Remaining time is around something like 65 and a power shutdown and restart is happening again the Remaining in PulseTime1 is set back to the 180. But I am expecting the remaining time to be continue from where it was(65 second) before the power shutdown.

I know that with mqtt or some other connection we can maintain this PulseTime state but I want a standalone solution to tasmota for my scenerio without any external system.

Can someone please suggest me a solution or fix for this please.

Noschvie, 2025-06-09:
Have a look to the docs

https://tasmota.github.io/docs/Commands/#pulsetime

Arun-R-S, 2025-06-09:
Hi @Noschvie, I have seen the document and i was understart about the pulseTime. 

But what I want is a solution for my scenerio. If i set the PulseTime to turn on a switch only for 680 seconds it should turn on only for this seconds even the tasmota(esp32) is power off and restart multiple times.

But what really happen is whenever the tasmota device is power off and restart. the Remaining time is actually set back to 680 again.

So in a case if i set 3600 seconds for PulseTime then there is a frequent power off means then the switch won't turn of ever.

This is the issue i am facing.

sfromis, 2025-06-09:
You have no "issue" (in the Github/Tasmota sense), you "just" want something which is not supported, `PulseTime` is not even meant to persist across Tasmota restarts.

If you want something "surviving" unstable power, you'd have to do something "less straightforward" - possibly including calculating the time you want it to turn off, and have a rule to turn off when that time is reached. Details depend on what you exactly want of timing around the bad power situation.