arendst/Tasmota #23536: Autoconf issue since 14.6.0

Grumium, 2025-06-11:
Hi,
the recent version breaks autoconf functionality for me.

```
18:47:26.622 CFG: removing autoconf files
18:47:26.633 CFG: downloading 'https://raw.githubusercontent.com/tasmota/autoconf/main/esp32/M5Stack_Core2.autoconf'
18:47:27.250 HTP: Connected in 614 ms, stack low mark 17796
18:47:27.445 HTP: Restart
18:47:27.446 SRC: WebGui from 10.42.0.1
18:47:28.647 APP: Restarting
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:268
ho 0 tail 12 room 4
load:0x40078000,len:13248
load:0x40080400,len:4
load:0x40080404,len:2928
entry 0x4008055c

00:00:00.002 HDW: ESP32-D0WDQ6-V3 v3.1 (PSRAM)
00:00:00.019 UFS: FlashFS mounted with 12580 kB free
00:00:00.029 CFG: Loaded from File, Count 27
00:00:00.037 QPC: Count 1
00:00:00.038 CFG: CR 392/699, Busy 0
00:00:00.040 SPI: Bus1 using GPIO18(CLK), GPIO23(MOSI) and GPIO38(MISO)
00:00:00.041 I2C: Bus1 using GPIO33(SCL) and GPIO32(SDA)
00:00:00.051 I2C: Bus2 using GPIO22(SCL) and GPIO21(SDA)
00:00:00.656 I2C: BM8563 found at 0x51 (bus2)
00:00:00.658 RTC: UTC 2025-06-11T17:47:30Z, DST 2025-03-30T02:00:00, STD 2025-10-26T03:00:00
18:47:30.000 RTC: Synced by BM8563
18:47:30.015 TFS: File 'mcp23x.dat' not found
18:47:30.015 MCP: Invalid template
18:47:30.016 TYA: Active=0
18:47:30.018 ROT: Mode 1
18:47:30.024 BRY: GC from 3832 to 3632 bytes, objects freed 4/21 (in 0 ms) - slots from 37/61 to 29/61
18:47:30.057 BRY: GC from 6949 to 5256 bytes, objects freed 9/47 (in 1 ms) - slots from 74/91 to 50/91
18:47:30.058 BRY: Berry initialized, RAM used 5256 bytes
18:47:30.077 CFG: loading M5Stack_Core2.autoconf#preinit.be
18:47:30.200 BRY: GC from 10529 to 9036 bytes, objects freed 19/105 (in 1 ms) - slots from 140/152 to 106/152
18:47:30.207 I2C: AXP192 detected on bus 2
18:47:30.148 CFG: loaded  M5Stack_Core2.autoconf#preinit.be
18:47:30.154 BRY: No 'preinit.be'
18:47:30.167 DSP: File descriptor used
18:47:30.815 DSP: ILI9342 initialized
18:47:30.819 SRC: Restart
18:47:30.825 Project tasmota - Tasmota Version 14.5.0(release-lvgl-haspmota)-3_1_1(2025-02-18T14:37:15)
18:47:30.827 LVG: Allocating buffer1 37 KB in main memory (flushlines 60)
18:47:30.844 LVG: LVGL initialized
18:47:30.881 LVG: Refreshed 19200 pixels in 10 ms (1920 pix/ms)
18:47:30.897 LVG: Refreshed 19200 pixels in 10 ms (1920 pix/ms)
18:47:30.913 LVG: Refreshed 19200 pixels in 9 ms (2133 pix/ms)
18:47:30.925 LVG: Refreshed 19200 pixels in 10 ms (1920 pix/ms)
18:47:30.929 ETH: No ETH MDC and ETH MDIO GPIO defined
18:47:30.976 SRC: Berry
18:47:30.977 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 163, Pld -99, Data '{"NAME":"M5Stack Core 2","GPIO":[7840,1,7776,0,6720,768,0,0,7808,1,1,800,0,0,736,1,0,641,609,704,0,0,0,1,6210,0,0,0,640,608,7872,1,1,0,672,7968],"FLAG":0,"BASE":1}'
18:47:30.999 CFG: CR 392/699, Busy 0
18:47:31.002 RSL: RESULT = {"NAME":"M5Stack Core 2","GPIO":[7840,1,7776,0,6720,768,0,0,7808,1,1,800,0,0,736,1,0,641,609,704,0,0,0,1,6210,0,0,0,640,608,7872,1,1,0,672,7968],"FLAG":0,"BASE":1}
18:47:31.016 SRC: Berry
18:47:31.017 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 1, Pld 0, Data '0'
18:47:31.018 RSL: RESULT = {"Module":{"0":"M5Stack Core 2"}}
18:47:31.030 CFG: 'init.bat' done, restarting
18:47:31.051 BRY: No 'autoexec.be'
18:47:31.205 CFG: Saved, Count 28, Bytes 4096
18:47:31.350 WIF: Checking connection...
18:47:31.351 WIF: Attempting connection...
18:47:32.001 WIF: Connecting to AP1 22167-T14G3 Channel 11 BSSId 00:A5:54:B7:AB:2E in mode HT40 as tasmota-9F1AEC-6892...
18:47:32.061 WIF: IPv4 10.42.0.82, mask 255.255.255.0, gateway 10.42.0.1
18:47:33.788 APP: Restarting
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:268
ho 0 tail 12 room 4
load:0x40078000,len:13248
load:0x40080400,len:4
load:0x40080404,len:2928
entry 0x4008055c

00:00:00.001 HDW: ESP32-D0WDQ6-V3 v3.1 (PSRAM)
00:00:00.029 UFS: FlashFS mounted with 12580 kB free
00:00:00.042 CFG: Loaded from File, Count 28
00:00:00.044 FRC: Some settings have been reset (2)
00:00:00.044 CFG: CR 392/699, Busy 0
00:00:00.047 SPI: Bus1 using GPIO18(CLK), GPIO23(MOSI) and GPIO38(MISO)
00:00:00.058 I2C: Bus1 using GPIO33(SCL) and GPIO32(SDA)
00:00:00.058 I2C: Bus2 using GPIO22(SCL) and GPIO21(SDA)
00:00:00.673 I2C: BM8563 found at 0x51 (bus2)
00:00:00.675 RTC: UTC 2025-06-11T17:47:35Z, DST 2025-03-30T02:00:00, STD 2025-10-26T03:00:00
18:47:35.000 RTC: Synced by BM8563
18:47:35.015 TFS: File 'mcp23x.dat' not found
18:47:35.016 MCP: Invalid template
18:47:35.016 TYA: Active=0
18:47:35.018 ROT: Mode 1
18:47:35.024 BRY: GC from 3832 to 3632 bytes, objects freed 4/21 (in 0 ms) - slots from 37/61 to 29/61
18:47:35.064 BRY: GC from 6965 to 5256 bytes, objects freed 9/47 (in 1 ms) - slots from 73/91 to 50/91
18:47:35.065 BRY: Berry initialized, RAM used 5256 bytes
18:47:35.092 CFG: loading M5Stack_Core2.autoconf#preinit.be
18:47:35.328 BRY: GC from 10529 to 9036 bytes, objects freed 19/105 (in 0 ms) - slots from 140/152 to 106/152
18:47:35.334 I2C: AXP192 detected on bus 2
18:47:35.285 CFG: loaded  M5Stack_Core2.autoconf#preinit.be
(...)
```

s-hadinger, 2025-06-11:
Fix underway

Grumium, 2025-06-11:
Many thanks. Much appreciated!

closed 2025-06-11 by Grumium

s-hadinger, 2025-06-11:
Should be fixed now in development branch

Grumium, 2025-06-12:
Thank you @s-hadinger. Confirmed working.