arendst/Tasmota #23554: Energy display doubles after update

RonnyB1978, 2025-06-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [x ] Device used (e.g., Sonoff Basic): MOUS A1T
- [ ] Tasmota binary firmware version number used: 15.0.1
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: web based
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

I updated all my NOUS A1T and Shelly Plug S (with Tasmota) to version 15.0.1. I have seen, that all power values are doubled. Solar Pump before update (14.2.0) 80 watts, after update to version 15.0.0 or 15.0.1 167 watts.

Can anybody confirm this issue? I don't know if the update to 14.3.0 or 15.0.x is the issue?

Thanks yor help.

Ronny

[image]

helmar74, 2025-06-16:
I can't confirm this issue. I updated from 14.6.0 to (first 15.0.0) and then to 15.0.1.
Here 2 screenshot of a NOUS 1AT and a Blitzwoif SHP2 (on both a 100W bulb is connected):

[image]

[image]

RonnyB1978, 2025-06-16:
Hello Helmar.

Thanks for your feedback. I get normal values, when I goes back to 14.2.0. My Shelly Plug S with tasmota 15.0.x shows at my washing machine 4500 watts. The plug works only until 2500 watts in max. 😉

The plug at my solar pump shows 167 watts -> according to manual the pump has a maximum power consumption of 120 watts. I also have a separate Shelly 3EM. I will check double.

Thanks a lot.
Ronny

helmar74, 2025-06-16:
For my dishwasher it looks like < 15.0.x:

[image]

RonnyB1978, 2025-06-16:
Can you test it with firmware 14.2.0? I can understand it with many devices. The greater the consumption, the greater the difference. Perhaps you also have a plug where you can simulate something in particular. A vacuum cleaner, for example. Once with firmware 14.2.0 and once with firmware 15.0.0+.

RonnyB1978, 2025-06-21:
Hello Helmar.

Sorry, for the delay in submit of my tests. My son had his graduation ball. In the first picture you can see, that I use a Philips vacuum cleaner with max power cosumption of 900W.

[image]

Firmware 14.2.0 works perfect > see in my first video.

https://github.com/user-attachments/assets/cfd6de8d-8a93-498a-99a8-4374289cdc4c

When I upgrade to 14.3.0 or higher, it shows a power of approx. 1900 watts. Thats impossible. 😉

14.3.0
https://github.com/user-attachments/assets/54deeb88-0369-4c8c-b446-d490de15deb8

15.0.1
https://github.com/user-attachments/assets/ea58ef57-d444-40ec-bb45-fc052fa59dfd

and 15.0.1.1
https://github.com/user-attachments/assets/7d1082b3-e57b-42f9-ba7a-42942c747a34

 Thank you very much for help. Until then I will only use 14.2.0, which works.

Have a nice weekend and please stay healthy.

Ronny

Noschvie, 2025-06-21:
It would be helpful to find out the commit that leads to the break.

Noschvie, 2025-06-21:
When I look at the release notes of version 14.3.0, these points stand out:

- Support for BL0906 up to 6 channel energy monitor as used in Athom EM2/EM6 [#22167](https://github.com/arendst/Tasmota/issues/22167)
- Support for Sonoff SPM v1.3.0 [#13447](https://github.com/arendst/Tasmota/issues/13447)
- Energy command PowerSet 60,230 to calibrate both Current and Power with known resistive load of 60W at 230V using calibrated Voltage
- Energy command CurrentSet 60,230 to calibrate both Power and Current with known resistive load of 60W at 230V using calibrated Voltage
- Energy Log level 4 message when (Calculated) Apparent Power is less than Active Power indicating wrong calibration [#20653](https://github.com/arendst/Tasmota/issues/20653)

FerFemNemBem, 2025-06-25:
Just an Idea: as you noted, the "Shelly Plug S" which can be eassily used for testing - I would suggest doing very basic testing without using 3rd party tools for graphs or history:

1.) flash with tasmota 14.2.0, test with resistive load (e.g. light bulb 100W, not inductive or capacitive load) check using tasmota web interface or console (not using other tools like Grafana etc.)
2.) flash with tasmota 15.0.1, test with resistive load (same as in 1.) check using tasmota web interface or console (not using other tools like Grafana etc.)
3.) do a recalibration with tasmota 15.0.1 with resistive load (same as in 1.) check using tasmota web interface or console (not using other tools like Grafana etc.)
4.) flashing again tasmota 14.2.0 with resistive load (same as in 1.) check using tasmota web interface or console (not using other tools like Grafana etc.)

FerFemNemBem, 2025-06-25:
I think a toaster or kettle should work too and most poeple have something like this at home.
For calibration, I use another, already calibrated device between the wall socket and the tasmota device to be calibrated to doublecheck if the calibration was successful.
I wouldn't trust the values on the stickers on the devices.

closed 2025-08-09 by github-actions[bot]