arendst/Tasmota #23556: Invalid file signature

telizynav, 2025-06-16:
Whats wrong?

[image]

sfromis, 2025-06-16:
Hard to guess, as you are not telling what you did to get there, or what's in the log.

telizynav, 2025-06-16:
[image]

telizynav, 2025-06-16:
Console log:

09:23:15.211 UPL: File ncp-uart-nsw_6.7.9_115200.ota (192240 bytes)
09:23:15.332 UPP: Multicast (re)joined

sfromis, 2025-06-17:
Then you are not doing a valid attempt to update the firmware. What you posted first looked ok, but not the file names you posted now.
You can find the tested files here:
https://github.com/arendst/Tasmota/tree/development/tools/fw_SonoffZigbeeBridge_ezsp

BTW. It is not a good plan to just say "nothing happens", the log will have some amount of messages.