arendst/Tasmota #23613: Matter device state not working on tuya matter hub

vampywiz17, 2025-06-28:
### PROBLEM DESCRIPTION

I created a 4 relay board with matter  relay device (also try light device). My problem is that if set on it on tuya app, seems not change it to "on" in tuya app. I also open a thread these theme in tuya forum: 

https://www.tuyaos.com/viewtopic.php?t=4364

We talking about a lot, they send two firmware update that solve the problem , but not help at all. At the end, they said these:

> The original problem has been resolved. From the latest log, we found that the gateway subscribed to the devices in a full subscription manner, but the devices did not report all the subscriptions. As a result, after control, the devices that failed to subscribe could not report the switch status. This suspicion is more likely to be a problem on the device side.
thanks

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [x] Device used (e.g., Sonoff Basic): LilyGO TTGO 4 Channel ESP32 Relay Board (T-Relay)
- [x ] Tasmota binary firmware version number used: Tasmota 15.0.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used:  https://tasmota.github.io/install/
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
20:10:02.209 RSL: RESULT = {"NAME":"T-Relay ESP32","GPIO":[0,0,1,0,1,227,0,0,1,1,1,1,0,0,226,225,0,224,1,1,0,544,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
20:10:02.435 RSL: RESULT = {"Module":{"0":"T-Relay ESP32"}}
20:10:02.640 RSL: RESULT = {"GPIO0":{"Nincs":0},"GPIO1":{"Nincs":0},"GPIO2":{"Nincs":0},"GPIO3":{"Nincs":0},"GPIO4":{"Nincs":0},"GPIO5":{"Relé4":227},"GPIO6":{"Nincs":0},"GPIO7":{"Nincs":0},"GPIO8":{"Nincs":0},"GPIO9":{"Nincs":0},"GPIO10":{"Nincs":0},"GPIO11":{"Nincs":0},"GPIO12":{"Nincs":0},"GPIO13":{"Nincs":0},"GPIO14":{"Nincs":0},"GPIO15":{"Nincs":0},"GPIO16":{"Nincs":0},"GPIO17":{"Nincs":0},"GPIO18":{"Relé3":226},"GPIO19":{"Relé2":225},"GPIO20":{"Nincs":0},"GPIO21":{"Relé1":224},"GPIO22":{"Nincs":0},"GPIO23":{"Nincs":0},"GPIO24":{"Nincs":0},"GPIO25":{"LedLink":544},"GPIO26":{"Nincs":0},"GPIO27":{"Nincs":0},"GPIO32":{"Nincs":0},"GPIO33":{"Nincs":0},"GPIO34":{"Nincs":0},"GPIO35":{"Nincs":0},"GPIO36":{"Nincs":0},"GPIO37":{"Nincs":0},"GPIO38":{"Nincs":0},"GPIO39":{"Nincs":0}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
20:10:49.667 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","","",""],"Topic":"tasmota_015360","ButtonTopic":"0","Power":"1110","PowerLock":"0000","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
20:10:49.672 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32-HU.bin","RestartReason":"Software reset CPU","Uptime":"0T00:12:35","StartupUTC":"2025-06-28T18:58:14","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2025-06-28T13:11:11","SaveCount":67}}
20:10:49.676 RSL: STATUS2 = {"StatusFWR":{"Version":"15.0.1(release-tasmota32)","BuildDateTime":"2025.06.14 10:42:19","Core":"3_1_3","SDK":"5.3.3.250501","CpuFrequency":160,"Hardware":"ESP32-D0WD v1.0","CR":"395/699"}}
20:10:49.679 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"FileLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["iot_network",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000020"]}}
20:10:49.692 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":2047,"Free":832,"Heap":130,"StackLowMark":3,"PsrMax":4096,"PsrFree":4091,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DIO","Features":["040E","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80F1","00000814"],"Drivers":"1,2,!3,!4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,!73,!75,82,!86,!87,!88,!91,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82,89"}}
20:10:49.700 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-015360-4960","IPAddress":"192.168.31.206","Gateway":"192.168.31.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.31.1","DNSServer2":"0.0.0.0","Mac":"8C:4B:14:01:53:60","IP6Global":"fd7d:a578:b2c9:8ae7:8e4b:14ff:fe01:5360","IP6Local":"fe80::8e4b:14ff:fe01:5360%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.31.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
20:10:49.704 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_015360","MqttUser":"DVES_USER","MqttCount":0,"MqttTLS":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
20:10:49.709 RSL: STATUS7 = {"StatusTIM":{"UTC":"2025-06-28T19:10:49Z","Local":"2025-06-28T20:10:49","StartDST":"2025-03-30T02:00:00","EndDST":"2025-10-26T03:00:00","Timezone":"+01:00","Sunrise":"04:49","Sunset":"20:57"}}
20:10:49.715 RSL: STATUS10 = {"StatusSNS":{"Time":"2025-06-28T20:10:49"}}
20:10:49.722 RSL: STATUS11 = {"StatusSTS":{"Time":"2025-06-28T20:10:49","Uptime":"0T00:12:35","UptimeSec":755,"Heap":128,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"Berry":{"HeapUsed":18,"Objects":226},"POWER1":"OFF","POWER2":"ON","POWER3":"ON","POWER4":"ON","Wifi":{"AP":1,"SSId":"iot_network","BSSId":"E6:63:DA:65:1A:7A","Channel":6,"Mode":"HT40","RSSI":100,"Signal":-42,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
- Create a matter device (relay or light)
- Add it to Tuya app (include matter gateway)
- Try to control it

### EXPECTED BEHAVIOUR
 Possible to on and also off the device.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
I try it with HA matter , with these, it working well.

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2025-06-30:
I really need full logs to understand. Set `seriallog 3`or `weblog 3` (loglevel 4 is too verbose).

I need the logs at boot time, and when the app connects to the device and asks for subscription.

vampywiz17, 2025-07-01:
@s-hadinger 

reg log:

```
00:00:00.002 HDW: ESP32-D0WD v1.0 (PSRAM disabled)
00:00:00.002 HDW: PSRAM is disabled, requires specific compilation on this hardware (see doc)
00:00:00.039 UFS: FlashFS mounted with 304 kB free
00:00:00.056 CFG: Loaded from File, Szám 133
00:00:00.064 QPC: Count 1
00:00:00.065 CFG: CR 395/699, Busy 0
00:00:00.069 TYA: Active=0
00:00:00.074 ROT: Mode 1
00:00:00.102 CFG: No '*.autoconf' file found
00:00:00.130 MTR: Load_config = {'14': {'relay': 4, 'type': 'light0'}, '13': {'relay': 3, 'type': 'light0'}, '12': {'relay': 2, 'type': 'light0'}, '11': {'relay': 1, 'type': 'light0'}}
00:00:00.143 MTR: Loaded 0 fabric(s)
00:00:00.212 MTR: Manual pairing code: 36448116163
00:00:00.217 RSL: RESULT = {"Matter":{"Commissioning":1,"PairingCode":"36448116163","QRCode":"MT:Y.K90SO527WK1T3AL00"}}
00:00:00.538 BRY: Berry initialized, RAM used 10229 bytes
00:00:00.545 BRY: No 'preinit.be'
00:00:00.546 SRC: Restart
00:00:00.551 Projekt tasmota - Tasmota Verzió 15.0.1(release-tasmota32)-3_1_3(2025.06.14 10:42:19)
00:00:00.579 BRY: No 'autoexec.be'
00:00:00.856 WIF: Csatlakozás...
00:00:02.000 WIF: AP-hoz csatlakozás:1 iot_network Channel 6 BSSId E6:63:DA:65:1A:7A mód: HT40 mint tasmota-015360-4960...
00:00:02.102 WIF: IPv4 192.168.31.206, mask 255.255.255.0, gateway 192.168.31.1
00:00:03.259 MTR: +Add_Event (CRIT      6001) [00]0028/00 (StartUp) - 15.0.1.0
00:00:03.264 MTR: +Add_Event (CRIT      6002) [00]0033/03 (BootReason) - 1U
00:00:03.265 MTR: Configuring endpoints
00:00:03.276 MTR:   endpoint =     0 type:root
00:00:03.277 MTR:   endpoint =     1 type:aggregator
00:00:03.280 MTR:   endpoint =    11 type:light0 relay:1
00:00:03.283 MTR:   endpoint =    12 type:light0 relay:2
00:00:03.297 MTR:   endpoint =    13 type:light0 relay:3
00:00:03.300 MTR:   endpoint =    14 type:light0 relay:4
00:00:03.301 RSL: RESULT = {"Matter":{"Initialized":1}}
00:00:03.303 MTR: Starting UDP server on port: 5540
00:00:03.326 MTR: start mDNS on wifi host '8C4B14015360.local'
00:00:03.334 MTR: starting mDNS on wifi 'D2572F2264A272BC' ptr to `8C4B14015360.local`
00:00:03.334 MTR: adding subtype: _L3879
00:00:03.335 MTR: adding subtype: _S15
00:00:03.336 MTR: adding subtype: _V65521
00:00:03.336 MTR: adding subtype: _CM1
00:00:03.992 WIF: IPv6 Local fe80::8e4b:14ff:fe01:5360%st1
00:00:04.106 WIF: Csatlakoztatva
00:00:04.365 HTP: Web server active on tasmota-015360-4960 with IP address 192.168.31.206
00:00:05.037 RTC: UTC 2025-07-01T16:43:26Z, nyári idő 2025-03-30T02:00:00, STD 2025-10-26T03:00:00
17:43:26.000 RTC: Synced by NTP
17:43:26.313 RSL: INFO1 = {"Info1":{"Module":"T-Relay ESP32","Version":"15.0.1(release-tasmota32)","FallbackTopic":"cmnd/DVES_015360_fb/","GroupTopic":"cmnd/tasmotas/"}}
17:43:26.326 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-015360-4960","IPAddress":"192.168.31.206","IP6Global":"","IP6Local":"fe80::8e4b:14ff:fe01:5360%st1"}}
17:43:26.337 RSL: INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":10}}
17:43:26.349 RSL: RESULT = {"POWER1":"OFF"}
17:43:26.350 RSL: POWER1 = OFF
17:43:26.350 RSL: RESULT = {"POWER2":"OFF"}
17:43:26.361 RSL: POWER2 = OFF
17:43:26.362 RSL: RESULT = {"POWER3":"OFF"}
17:43:26.362 RSL: POWER3 = OFF
17:43:26.363 RSL: RESULT = {"POWER4":"OFF"}
17:43:26.374 RSL: POWER4 = OFF
17:43:28.477 HTP: Menü
17:43:29.064 QPC: Reset
17:43:30.925 APP: Újraindulások száma 10
17:43:30.931 RSL: STATE = {"Time":"2025-07-01T17:43:30","Uptime":"0T00:00:09","UptimeSec":9,"Heap":141,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"Berry":{"HeapUsed":11,"Objects":151},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"iot_network","BSSId":"E6:63:DA:65:1A:7A","Channel":6,"Mode":"HT40","RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:04"}}
17:43:32.018 CFG: Saved, Szám 134, Byte-ok 4096
17:43:38.136 HTP: Menü
17:43:41.286 HTP: Tools
17:43:43.258 HTP: Konzol
17:44:19.954 MTR: >Received  (     0) PBKDFParamRequest from [192.168.31.200]:57989
17:44:19.971 MTR: +Session   ( 44021) from '[192.168.31.200]:57989'
17:44:19.985 MTR: <Replied   (     0) PBKDFParamResponse
[... 100 lines cut ...]
17:44:34.363 MTR: >Command   ( 33197) [00]003E/0009 UpdateFabricLabel 
17:44:34.369 MTR: <Replied   ( 33197) [00]003E/0008 NOCResponse
17:44:34.504 MTR: >Read_Attr1( 33197) [0E]0039/0011 (Reachable) - 1
17:44:34.528 MTR: >Read_Attr1( 33197) [0C]0039/0011 (Reachable) - 1
17:44:34.553 MTR: >Read_Attr1( 33197) [0B]0039/0011 (Reachable) - 1
17:44:34.578 MTR: >Read_Attr1( 33197) [0D]0039/0011 (Reachable) - 1
17:44:34.624 MTR: >Subscribe ( 33197) [**]****/**** (min=0, max=60, keep=0) sub=56561 fabric_filtered=true
17:44:34.630 MTR: >Read_Attr ( 33197) [**]****/!****
17:44:34.632 MTR: >Read_Event( 33197) [**]****/** 
17:44:35.162 MTR: >Sub_OK    ( 33197) sub=56561
17:44:35.642 MTR: >Sub_OK    ( 33197) sub=56561
17:44:36.050 MTR: >Sub_OK    ( 33197) sub=56561
17:44:36.388 MTR: >Sub_OK    ( 33197) sub=56561
17:44:36.798 MTR: >Read_Attr ( 33197) [**]****/!****
17:44:36.800 MTR: >Read_Event( 33197) [**]****/** 
17:44:37.221 MTR: >Read_Attr ( 33197) [**]****/!****
17:44:37.224 MTR: >Read_Event( 33197) [**]****/** 
17:44:37.649 MTR: >Read_Attr ( 33197) [**]****/!****
17:44:37.652 MTR: >Read_Event( 33197) [**]****/** 
17:44:38.139 MTR: >Read_Attr ( 33197) [**]****/!****
17:44:38.144 MTR: >Read_Event( 33197) [**]****/** 
17:44:38.551 MTR: >Sub_OK    ( 33197) sub=56561
17:44:38.955 MTR: .          Duplicate encrypted message = 33845011 counter=33845012
17:44:38.960 MTR: .          Duplicate encrypted message = 33845009 counter=33845012
```

s-hadinger, 2025-07-06:
Can you paste a copy of your file `_matter_device.json`?

s-hadinger, 2025-07-06:
I just tried with Tuya Smart on iOS, version 6.7.0.

I did observe a freeze of Tasmota for 50 seconds which I can't explain yet. But eventually it stabilized, and all 4 relays work. They can be changed from the UI and from the Tuya app, with real-time update both ways.

vampywiz17, 2025-07-06:
@s-hadinger 

Hmm ... are you use matter gateway or only the tuya app?

s-hadinger, 2025-07-06:
Ok, I'll test further. I'm using the Tuya Smart App alone

vampywiz17, 2025-07-06:
@s-hadinger 

Now i delete gateway. and add again tasmota. Seems it working. it property report the state via tuya app.

So seems the problem is the gateway.

vampywiz17, 2025-08-09:
only if i use tuya app only. If i use gateway, it not stabilize...

vampywiz17, 2025-08-22:
@s-hadinger 

Any log or something that i able to help you? 

I try other devide (set it up a roller shutter on Tasmota) and same problem... When the device communicate with app, it work well... but if the communication are adding to tuya matter hub , the communication will be one way... send the command but not receive it.

With my test, seems Tuya matter connection work with:

- 1. Connect matter device to Tuya app, if no matter hub in network nothing happen the device work
- 2.  After the matter device - Tuya app connected successfully and Tuya app found tuya matter hub in network, some second later, it will migrate the connection to the Hub. (in these case if you whach the app interface, that you see the matter device dissappear on tuya interface and some sec later, appear again. it is the migration process i think)

vampywiz17, 2025-09-10:
hello @s-hadinger 

Today i play a bit and use these node on node-RED to make "virtual" Matter devices: https://github.com/sammachin/node-red-matter-bridge

I connect these virtual gateway to my Tuya matter hub and it work well. I able to control the device that i config it with NodeRED.

So if you have some time , check these repo, what the guy do different.