arendst/Tasmota #23661: Warning: uv command not found

tbretz65, 2025-07-10:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
fresh download from github

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
no rules

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Processing tasmota32 (framework: arduino; platform: https://github.com/tasmota/platform-espressif32/releases/download/2025.07.30/platform-espressif32.zip; board: esp32)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Nothing more to solidify
-DCONFIG_TASMOTA_FLASHMODE_DIO
*** use provided user_config_override.h as planned ***
*** use provided platformio_override.ini as planned ***
*** use provided platformio_tasmota_cenv.ini as planned ***
Warning: uv command not found
Error: uv command not found
FileNotFoundError: [Errno 2] No such file or directory: 'uv':
  File "/root/.platformio/penv/lib/python3.9/site-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "/root/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/root/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/root/.platformio/platforms/espressif32/builder/main.py", line 205:
    install_esptool(env)
  File "/root/.platformio/platforms/espressif32/builder/main.py", line 191:
    subprocess.check_call([
  File "/usr/lib/python3.9/subprocess.py", line 368:
    retcode = call(*popenargs, **kwargs)
  File "/usr/lib/python3.9/subprocess.py", line 349:
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.9/subprocess.py", line 951:
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.9/subprocess.py", line 1823:
    raise child_exception_type(errno_num, err_msg, err_filename)
==================================================================================== [FAILED] Took 1.29 seconds ====================================================================================

Environment    Status    Duration
-------------  --------  ------------
tasmota32      FAILED    00:00:01.286
=============================================================================== 1 failed, 0 succeeded in 00:00:01.286 ===============================================================================

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
download tasmota from github
run "pio run -e tasmota32" ( only with tasmota32-xx )

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2025-07-10:
I have the problem in docker-tasmota and I had to add 'uv' to pip install in the dockerfile
But I have no problem with vscode in windows

I never understood how dependencies are installed in platformio
I can see that 'uv' is installed in github workflows but can't find any requirement to have it installed by platformio so I don't know how it gets into my windows' platformio venv

closed 2025-07-10 by tbretz65

barbudor, 2025-07-10:
So you installed platformio in your venv without installing uv ?

In windows + vscode, the vscode extension creates a venv with python 3.10

But I still don't get where it knows to install uv

@Jason2866 could you kindly explain how dependencies are managed ?