arendst/Tasmota #23711: mi32cfg command unknown

cph25, 2025-07-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I used TamoCompiler v12.7.0 creating a firmware.bin for ESP32s3 with 
#ifdef USE_BLE_ESP32
  #undef USE_BLE_ESP32
#endif
#define USE_BLE_ESP32

#ifdef USE_MI_ESP32
  #undef USE_MI_ESP32
#endif
#define USE_MI_ESP32
in user_config_override.h. 
Tasmota Version 15.0.1(TasmoCompiler-esp32s3)-3_1_3
Works fine and lists the Miflora sensors in the main menu.
According to the MI32 legacy section of the Tasmota documentation v15.0.1 (Homeassistant and Tasmota - BLE sensors) creating a mi32cfg file is a prerequisite for the disco.be berry script. Executing the command “mi32cfg” in the Tasmota console results in {"Command":"Unknown","Input":"MI32CFG"}. Running “Mi32Option1” works fine, but “MI32Option3“ results in {"Command":"Error","Input":"MI32OPTION3"}. 
I studied the documentation in order to find a spec of the JSON for the mi32cfg to create it manually. Unfortunately I didn’t find one. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [ ] Device used (e.g., Sonoff Basic): esp32s3_____
- [ ] Tasmota binary firmware version number used: 15.0.1_____
  - [ ] Pre-compiled
  - [ x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:05:53.721 MQT: stat/tasmota_4DCA9C/RESULT = {"NAME":"ESP32S3","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,0,0,0,0,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":1}
15:05:53.964 MQT: stat/tasmota_4DCA9C/RESULT = {"Module":{"1":"ESP32S3"}}
15:05:54.216 MQT: stat/tasmota_4DCA9C/RESULT = {"GPIO0":{"None":0},"GPIO1":{"None":0},"GPIO2":{"None":0},"GPIO3":{"None":0},"GPIO4":{"None":0},"GPIO5":{"None":0},"GPIO6":{"None":0},"GPIO7":{"None":0},"GPIO8":{"ADC Input1":4704},"GPIO9":{"None":0},"GPIO10":{"None":0},"GPIO11":{"None":0},"GPIO12":{"None":0},"GPIO13":{"None":0},"GPIO14":{"None":0},"GPIO15":{"None":0},"GPIO16":{"None":0},"GPIO17":{"None":0},"GPIO18":{"None":0},"GPIO19":{"None":0},"GPIO20":{"None":0},"GPIO21":{"None":0},"GPIO33":{"None":0},"GPIO34":{"None":0},"GPIO35":{"None":0},"GPIO36":{"None":0},"GPIO37":{"None":0},"GPIO38":{"None":0},"GPIO39":{"None":0},"GPIO40":{"None":0},"GPIO41":{"None":0},"GPIO42":{"None":0},"GPIO43":{"None":0},"GPIO44":{"None":0},"GPIO45":{"None":0},"GPIO46":{"None":0},"GPIO47":{"None":0},"GPIO48":{"None":0}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
15:07:22.214 MQT: stat/tasmota_4DCA9C/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_4DCA9C","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
15:07:22.222 MQT: stat/tasmota_4DCA9C/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":" ","RestartReason":"Software reset CPU","Uptime":"0T00:30:25","StartupUTC":"2025-07-22T13:36:57","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2025-07-21T21:38:22","SaveCount":19}}
15:07:22.227 MQT: stat/tasmota_4DCA9C/STATUS2 = {"StatusFWR":{"Version":"15.0.1(TasmoCompiler-esp32s3)","BuildDateTime":"2025-07-22T11:39:29","Core":"3_1_3","SDK":"5.3.3.250501","CpuFrequency":240,"Hardware":"ESP32-S3 v0.2","CR":"405/699"}}
15:07:22.233 MQT: stat/tasmota_4DCA9C/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"FileLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Peter IoT Netzwerk",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004002","00000000"]}}
15:07:22.248 MQT: stat/tasmota_4DCA9C/STATUS4 = {"StatusMEM":{"ProgramSize":1519,"Free":1360,"Heap":144,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":16384,"FlashSize":16384,"FlashChipId":"184068","FlashFrequency":80,"FlashMode":"QIO","Features":["0809","0F4045CE","00040001","60020FC3","180C0000","D0320880","00092088","00200000","54000048","40000080","00000004"],"Drivers":"1,2,!4,9,10,12,26,29,!35,38,50,52,62,!68,79,!121","Sensors":"1,2,5,6,7,8,9,14,26,27,38,39,46,47,52,55,58,59,62,63,65,72,73,74,88,127","I2CDriver":"8,9,10,15,20,21,32,36,41,42,43,45,51,52,60"}}
15:07:22.255 MQT: stat/tasmota_4DCA9C/STATUS5 = {"StatusNET":{"Hostname":"tasmota-4DCA9C-2716","IPAddress":"192.168.178.26","Gateway":"192.168.178.1","Subnetmask":"255.255.0.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"removed","IP6Global":"","IP6Local":"fe80::1220:baff:fe4d:ca9c%st1","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":19.0}}
15:07:22.263 MQT: stat/tasmota_4DCA9C/STATUS6 = {"StatusMQT":{"MqttHost":"rasp-pi-2.fritz.box","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4DCA9C","MqttUser":"DVES_USER","MqttCount":1,"MqttTLS":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
15:11:06.753 CMD: mi32cfg
15:11:06.754 SRC: WebConsole from 192.168.178.61
15:11:06.755 CMD: Grp 0, Cmd 'MI32CFG', Idx 1, Len 0, Pld -99, Data ''
15:11:06.767 MQT: stat/tasmota_4DCA9C/RESULT = {"Command":"Unknown","Input":"MI32CFG"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
mi32cfg should create the mi32cfg file

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Staars, 2025-07-22:
I have no experience with Tasmocompiler, but the internal macro definition to select the legacy driver is:
#define USE_MI_EXT_GUI

Thus you may try to add it to the lines from your first post.

Staars, 2025-07-22:
That is definitely not the legacy driver.
I will take a look at Tasmocompiler.
Alternatively you can use ready made binaries from here:
https://jason2866.github.io/Tasmota-specials/
(Legacy is MI BLE)

cph25, 2025-07-23:
Thank you, looks great and I know now, what the MIFlore dashboard looks like.
But now BLEAlias disappeared
14:31:15.443 CMD: BLEAlias C47C8D69FFE6=Zitrone
14:31:15.457 MQT: stat/tasmota_4DCA9C/RESULT = {"Command":"Unknown","Input":"BLEALIAS C47C8D69FFE6=Zitrone"}