arendst/Tasmota #23720: Problem with NeoPool Controller and Tasmota 15

AndreasTrinkle, 2025-07-26:
### PROBLEM DESCRIPTION
Since the last version 15 of Tasmota, the NeoPool controller no longer works. On the one hand the RX and TX are obviously swapped, on the other hand (after swapping the inputs) only completely nonsensical values are displayed.
After downgrading to Tasmota 14.x, everything is stable again (with the old settings)

AndreasTrinkle, 2025-07-29:
I'm a bit embarrassed. I was actually already looking for the issue, but didn't find this thread. Thank you very much. I assume it will also be the solution to my problem.

curzon01, 2025-07-29:
Once you have resolved it, e.g., by reassigning the GPIOs, it would be helpful for others if you could briefly describe whether and how it was resolved. Please close this ticket afterwards.

edi0177, 2025-08-20:
Having the same problem but it did actually work with 14.6, but stopped working after i added some gui scripts.
Iam now back to 14.4 but would be interested in a solution. My Problem with 14.4 is, that the device disappears after a few days and only will come back after a power-off. Before with 14.6 it was running for weeks.

Device: ESP32 Dev Kit C V2