arendst/Tasmota #23790: WS2812 no work

bananajoe86, 2025-08-14:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I'm getting desperate trying to control a WS2812. I thought I could connect it to a GPIO and it would light up, but my strip doesn't do anything. To test its functionality, I installed WLED on the ESP32-D1, and I was able to control the strip properly via the GPIO22, so the hardware is fine. I installed Tasmota on the ESP32 again, declared the GPIO22 with WS2812...the strip doesn't do anything.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [x ] Device used (e.g., Sonoff Basic): Wemos D1 ESP32
- [x] Tasmota binary firmware version number used: _15.1____
  - [x ] Pre-compiled
  - [x ] Self-compiled
- [x ] Flashing tools used: __webinstaller___
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
13:40:28.692 CMD: Backlog Template; Module; GPIO 255
13:40:28.710 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
13:40:28.941 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
13:40:29.195 RSL: RESULT = {"GPIO0":{"None":0},"GPIO1":{"None":0},"GPIO2":{"None":0},"GPIO3":{"None":0},"GPIO4":{"None":0},"GPIO5":{"None":0},"GPIO6":{"None":0},"GPIO7":{"None":0},"GPIO8":{"None":0},"GPIO9":{"None":0},"GPIO10":{"None":0},"GPIO11":{"None":0},"GPIO12":{"None":0},"GPIO13":{"None":0},"GPIO14":{"None":0},"GPIO15":{"None":0},"GPIO16":{"None":0},"GPIO17":{"None":0},"GPIO18":{"None":0},"GPIO19":{"None":0},"GPIO20":{"None":0},"GPIO21":{"None":0},"GPIO22":{"WS28121":1376},"GPIO23":{"None":0},"GPIO24":{"None":0},"GPIO25":{"None":0},"GPIO26":{"None":0},"GPIO27":{"None":0},"GPIO32":{"None":0},"GPIO33":{"None":0},"GPIO34":{"None":0},"GPIO35":{"None":0},"GPIO36":{"None":0},"GPIO37":{"None":0},"GPIO38":{"None":0},"GPIO39":{"None":0}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
13:41:03.118 CMD: Status 0
13:41:03.121 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_4A8A44","ButtonTopic":"0","Power":"1","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
13:41:03.154 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32-DE.bin","RestartReason":"Software reset CPU","Uptime":"0T00:01:16","StartupUTC":"2025-08-14T12:39:47","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2025-08-14T11:04:25","SaveCount":28}}
13:41:03.187 RSL: STATUS2 = {"StatusFWR":{"Version":"15.0.1(release-tasmota32)","BuildDateTime":"2025.06.14 10:39:16","Core":"3_1_3","SDK":"5.3.3.250501","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.1","CR":"422/699"}}
13:41:03.209 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"FileLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ!7390_alt",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
13:41:03.240 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":2045,"Free":834,"Heap":155,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164068","FlashFrequency":40,"FlashMode":"DIO","Features":["0407","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80F1","00000814"],"Drivers":"1,2,!3,4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,!73,!75,82,!86,!87,!88,!91,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82,89"}}
13:41:03.305 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-4A8A44-2628","IPAddress":"192.168.178.46","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.103","DNSServer2":"fda6:af00:15a4:0:7642:7fff:fe06:a65c","Mac":"F4:65:0B:4A:8A:44","IP6Global":"2001:16b8:ca20:d300:f665:bff:fe4a:8a44","IP6Local":"fe80::f665:bff:fe4a:8a44%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.178.103","DNSServer2":"fda6:af00:15a4:0:7642:7fff:fe06:a65c","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
13:41:03.362 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4A8A44","MqttUser":"DVES_USER","MqttCount":0,"MqttTLS":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:41:03.377 RSL: STATUS7 = {"StatusTIM":{"UTC":"2025-08-14T12:41:03Z","Local":"2025-08-14T13:41:03","StartDST":"2025-03-30T02:00:00","EndDST":"2025-10-26T03:00:00","Timezone":"+01:00","Sunrise":"05:42","Sunset":"20:07"}}
13:41:03.403 RSL: STATUS10 = {"StatusSNS":{"Time":"2025-08-14T13:41:03"}}
13:41:03.410 RSL: STATUS11 = {"StatusSTS":{"Time":"2025-08-14T13:41:03","Uptime":"0T00:01:16","UptimeSec":76,"Heap":156,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":41},"POWER":"ON","Dimmer":63,"Color":"5023A1","HSBColor":"261,78,63","Channel":[31,14,63],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"FRITZ!7390_alt","BSSId":"74:42:7F:06:A6:5F","Channel":6,"Mode":"HT40","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
13:41:47.721 CMD: Grp 0, Cmd 'DIMMER', Idx 1, Len 3, Pld 100, Data '100'
13:41:47.724 RSL: RESULT = {"POWER":"ON","Dimmer":100,"Color":"7E38FF","HSBColor":"261,78,100","Channel":[49,22,100]}
13:41:48.466 CFG: Saved, Anzahl 30, Bytes 4096
13:41:49.322 HTP: Werkzeuge
13:41:50.142 HTP: Konsole
13:41:54.469 WIF: Prüfe Verbindung…

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

bananajoe86, 2025-08-14:
You´re right. 
Downgrading to 14.6 makes no difference.
Downgrading to 13.4, works as expected.

bananajoe86, 2025-08-15:
Thank you for checking. I don't have a special setup. Just normal 5V WS2812 with + connected to Vcc and - and GND (direct at Wemos). I have tried the data pin on different GPIOs. Currently on 22. If you need more information, please let me know.

sfromis, 2025-08-16:
Specs for WS2812 says that minimum signal voltage is 70% of VCC, which with 5V means 3.5V. One trick is to reduce the VCC voltage, like VCC of 4.5V would only require a signal of 3.15V. Of course, it is not a given that it *will* fail going below 70%, but it "should" not be expected to work outside of specs.

One trick used by some is to reduce voltage to the first pixel, and still connect full 5V to the rest. This can work because the first pixel would still produce an output signal matching it's VCC voltage.

s-hadinger, 2025-08-17:
Unfortunately this doesn't explain why it worked with previous versions. I suppose that you are using the default RMT configuration?

s-hadinger, 2025-08-17:
Thanks this is useful. I suppose you don't have access to any logic analyser. 

One thing that could also explain the problem could be a lack of internal pull-up. I can't remember how the ws2812 gpio is configured internally. That could also explain some behavior

sfromis, 2025-08-21:
For me, it also works with Tasmota 15.0.1.3 (latest dev) using an ESP32 module which looks very similar to the one in the image you posted 6 days ago. Same gpio 22 as you posted about using.