arendst/Tasmota #23874: Wireguard stays in failed state if DNS-Request fails

fleinze, 2025-09-05:
### PROBLEM DESCRIPTION
When wireguard is started (manually or automatically at startup) a dns-request is sent if the endpoint address requires it. If this dns-request fails or times out wireguard stays in a failed state until it is manually restarted with backlog `wgstop;wgconnect`
Other Services like NTP or MQTT retry after failed DNS Requests.
It would be a great to add dns retries to wireguard and it 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [ ] Device used (e.g., Sonoff Basic): ZB-GW03-V1.2
- [ ] Tasmota binary firmware version number used: 15.0.1.3
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: OTA Update
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
00:00:02.284 ETH: IPv4 10.15.20.156, mask 255.255.255.0, gateway 10.15.20.1
00:00:02.507 WIF: Checking connection...
00:00:02.507 WIF: Attempting connection...
00:00:03.141 WIF: Connecting to AP1 MuF24n Channel 6 BSSId C4:41:1E:36:D8:79 in mode HT40 as tasmota-zbtest...
00:00:04.002 ETH: IPv6 Local fe80::9654:c5ff:feee:5463
00:00:04.004 NTP: Sync time...
00:00:04.005 DNS: dns_gethostbyname_addrtype errg=-6 counter=1
00:00:04.005 WIF: DNS failed for 2.pool.ntp.org after 0 ms
00:00:04.015 mDN: Initialized 'tasmota-zbtest.local'
00:00:04.024 HTP: Web server active on tasmota-zbtest-eth.local with IP address 10.15.20.156
00:00:04.031 WG : Found available network interface: en1
00:00:04.083 DNS: dns_gethostbyname_addrtype errg=-6 counter=2
00:00:04.084 WG : hostByName failed 'heinze.at'
00:00:04.084 WG : wireguard_peer_lookup_by_peer_index: peer not found
00:00:04.084 WG : Failed to add allowed_ips, no space left
00:00:04.087 NTP: Sync time...
```
The DNS log was manually compiled and is not a standard feature.

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
Wireguard should automatically wait for a certain amount of time (e.g. 30s) restart after a failed DNS request.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2025-09-08:
You have probably issues since trying to use `Initialized 'tasmota-zbtest.local'` Using mDNS extension .local is **not** supported with Tasmota. It does not work and creates issues.

fleinze, 2025-10-04:
PR is still in open