arendst/Tasmota #23961: BLE stopps reconizing adverts after a few seconds

SteWers, 2025-09-28:
### PROBLEM DESCRIPTION
BLE stopps reconizing adverts after a few seconds. For testing I set `BLEMaxAge` to 180. It happens also with the default value of 600, but so I haven't to wait so long.

In the attached log you can see that the adverts increase until timestamp 12:32:06.167.
At 12:34:52.061 the seen devices will be removed, because the max age is reached.
120 seconds later (12:36:45.053) NimBLE became to restart.
After that the adverts increase for a short time.

I think this is related to PR #23553 as it is working until version 15.0.1 (release-bluetooth)
### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [x] Device used: ESP32-DevKit (ESP32-D0WDQ6 v1.0)
- [x] Tasmota binary firmware version number used: 15.0.1.4 (bluetooth) and before
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
{"NAME":"ESP32-DevKit","GPIO":[1,1,576,1,0,0,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
{"Module":{"0":"ESP32-DevKit"}}
{"GPIO0":{"None":0},"GPIO1":{"None":0},"GPIO2":{"LedLink_i":576},"GPIO3":{"None":0},"GPIO4":{"None":0},"GPIO5":{"None":0},"GPIO6":{"None":0},"GPIO7":{"None":0},"GPIO8":{"None":0},"GPIO9":{"None":0},"GPIO10":{"None":0},"GPIO11":{"None":0},"GPIO12":{"None":0},"GPIO13":{"None":0},"GPIO14":{"None":0},"GPIO15":{"None":0},"GPIO16":{"None":0},"GPIO17":{"None":0},"GPIO18":{"None":0},"GPIO19":{"None":0},"GPIO20":{"None":0},"GPIO21":{"None":0},"GPIO22":{"None":0},"GPIO23":{"None":0},"GPIO24":{"None":0},"GPIO25":{"None":0},"GPIO26":{"None":0},"GPIO27":{"None":0},"GPIO32":{"None":0},"GPIO33":{"None":0},"GPIO34":{"None":0},"GPIO35":{"None":0},"GPIO36":{"None":0},"GPIO37":{"None":0},"GPIO38":{"None":0},"GPIO39":{"None":0}}
```
- [x] Provide the output of this command: `Status 0`:
```lua
{"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_C7036C","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32-bluetooth.bin","RestartReason":"Software reset CPU","Uptime":"0T00:03:11","StartupUTC":"2025-09-28T13:03:47","Sleep":50,"CfgHolder":4617,"BootCount":152,"BCResetTime":"2024-11-02T18:18:17","SaveCount":236},"StatusFWR":{"Version":"15.0.1.4(bluetooth)","BuildDateTime":"2025.09.28 14:35:39","Core":"3_3_0","SDK":"5.3.4.250826","CpuFrequency":160,"Hardware":"ESP32-D0WDQ6 v1.0","CR":"440/699"},"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"FileLog":0,"SysLog":4,"LogHost":"10.0.0.107","LogPort":514,"SSId":["Metropole-F","MyHome-SIK"],"TelePeriod":10,"Resolution":"568180C0","SetOption":["0000C009","2805C80001000600003C5A0A192800000000","00008080","00006000","00004002","00008001"]},"StatusMEM":{"ProgramSize":1754,"Free":1125,"Heap":71,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":"DIO","Features":["0407","8F8AC7CF","00148001","000000CF","010013C0","C0000981","40004080","00200000","5400082C","40000080","00000004"],"Drivers":"1,2,!3,!4,!5,!8,9,10,12,!16,!20,!21,!24,26,!27,29,!35,38,50,52,!59,62,!68,79,82,85,!121","Sensors":"1,2,3,5,6,52,62,127","I2CDriver":"7"},"StatusNET":{"Hostname":"tasmota-C7036C-0876","IPAddress":"10.0.0.144","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer1":"10.0.0.1","DNSServer2":"0.0.0.0","Mac":"9C:9C:1F:C7:03:6C","IP6Global":"","IP6Local":"fe80::9e9c:1fff:fec7:36c%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"10.0.0.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0},"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C7036C","MqttUser":"DVES_USER","MqttCount":0,"MqttTLS":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2025-09-28T13:06:58Z","Local":"2025-09-28T15:06:58","StartDST":"2025-03-30T02:00:00","EndDST":"2025-10-26T03:00:00","Timezone":99,"Sunrise":"07:45","Sunset":"19:35"},"StatusSNS":{"Time":"2025-09-28T15:06:58","ESP32":{"Temperature":53.3},"TempUnit":"C"},"StatusSTS":{"Time":"2025-09-28T15:06:58","Uptime":"0T00:03:11","UptimeSec":191,"Heap":70,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":44},"Wifi":{"AP":2,"SSId":"MyHome-SIK","BSSId":"7C:FF:4D:01:B8:22","Channel":1,"Mode":"HT40","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:04"},"Hostname":"tasmota-C7036C-0876","IPAddress":"10.0.0.144"}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
12:31:50.577  10.0.0.144  local0.error   tasmota-C7036C-0876 tasmota - - - BLE: registerForAdvertismentCallbacks EQ3:401197f4
12:31:50.577  10.0.0.144  local0.error   tasmota-C7036C-0876 tasmota - - - BLE: registerForAdvertismentCallbacks iBeacon:4011e2d4
12:31:50.852  10.0.0.144  local0.error   tasmota-C7036C-0876 tasmota - - - BLE: registerForAdvertismentCallbacks MI32:40122b6c
12:31:50.863  10.0.0.144  local0.error   tasmota-C7036C-0876 tasmota - - - BLE: registerForScnCallbacks MI32:4011cc78
12:31:51.433  10.0.0.144  local0.error   tasmota-C7036C-0876 tasmota - - - BLE: MasterEnable->1
12:31:51.433  10.0.0.144  local0.error   tasmota-C7036C-0876 tasmota - - - BLE: BLETask:Starting NimBLE
12:31:56.168  10.0.0.144  local0.info    tasmota-C7036C-0876 tasmota - - - RSL: BLE = {"Time":"2025-09-28T12:31:56","BLE":{"scans":0,"adverts":27,"devices":3,"resets":0}}
12:32:06.167  10.0.0.144  local0.info    tasmota-C7036C-0876 tasmota - - - RSL: BLE = {"Time":"2025-09-28T12:32:06","BLE":{"scans":0,"adverts":42,"devices":3,"resets":0}}
[...]
12:34:46.157  10.0.0.144  local0.info    tasmota-C7036C-0876 tasmota - - - RSL: BLE = {"Time":"2025-09-28T12:34:46","BLE":{"scans":0,"adverts":42,"devices":3,"resets":0}}
12:34:52.061  10.0.0.144  local0.error   tasmota-C7036C-0876 tasmota - - - BLE: Delete device A4C138C549A3() by age lastseen 4 + maxage 180 < now 185.
12:34:52.074  10.0.0.144  local0.error   tasmota-C7036C-0876 tasmota - - - BLE: Deleted 1 devices
12:34:53.064  10.0.0.144  local0.error   tasmota-C7036C-0876 tasmota - - - BLE: Delete device 6077715CA607() by age lastseen 5 + maxage 180 < now 186.
12:34:53.064  10.0.0.144  local0.error   tasmota-C7036C-0876 tasmota - - - BLE: Deleted 1 devices
12:34:56.129  10.0.0.144  local0.info    tasmota-C7036C-0876 tasmota - - - RSL: BLE = {"Time":"2025-09-28T12:34:56","BLE":{"scans":0,"adverts":42,"devices":1,"resets":0}}
12:34:58.058  10.0.0.144  local0.error   tasmota-C7036C-0876 tasmota - - - BLE: Delete device D0D0030F7BF3() by age lastseen 10 + maxage 180 < now 191.
12:34:58.058  10.0.0.144  local0.error   tasmota-C7036C-0876 tasmota - - - BLE: Deleted 1 devices
12:35:06.135  10.0.0.144  local0.info    tasmota-C7036C-0876 tasmota - - - RSL: BLE = {"Time":"2025-09-28T12:35:06","BLE":{"scans":0,"adverts":42,"devices":0,"resets":0}}
[...]
12:36:36.144  10.0.0.144  local0.info    tasmota-C7036C-0876 tasmota - - - RSL: BLE = {"Time":"2025-09-28T12:36:36","BLE":{"scans":0,"adverts":42,"devices":0,"resets":0}}
12:36:45.053  10.0.0.144  local0.debug   tasmota-C7036C-0876 tasmota - - - BLE: scan stall? no adverts > 120s, restart BLE
12:36:45.159  10.0.0.144  local0.debug   tasmota-C7036C-0876 tasmota - - - BLE: BLETask: Restart NimBLE - restart Tasmota in 10 if not complt
[... cut ...]
12:36:56.164  10.0.0.144  local0.info    tasmota-C7036C-0876 tasmota - - - RSL: BLE = {"Time":"2025-09-28T12:36:56","BLE":{"scans":0,"adverts":75,"devices":3,"resets":1}}
[...]
12:38:46.110  10.0.0.144  local0.info    tasmota-C7036C-0876 tasmota - - - RSL: BLE = {"Time":"2025-09-28T12:38:45","BLE":{"scans":0,"adverts":75,"devices":3,"resets":1}}
```
### TO REPRODUCE
Take the last dev version and observe the BLE adverts and devices on main page. Also have a look at the age entries on the BLE config page.

### EXPECTED BEHAVIOUR
BLE devices should not disappear if they a still present and sending adverts.

Rushmed, 2025-10-18:
Got something Similar.
Since update from 15.0.1 to 15.1. BLE beacons in Tasmota range are continuosly connected and disconnected instead of staying connected.
Here is a Graph, the solid part was before the update and the interrupted part is on V 15.1.

[image]

Jason2866, 2025-10-19:
There are code changes in IDF and the NimBLE lib in Tasmota has been updated. Probably the regarding BLE part in Tasmota needs to be checked and adopted to this changes.
I know changes in the MI BLE code in Tasmota have been done.

sfromis, 2025-10-19:
What I noticed was that it happened after the PR to rename libs, was that also doing code changes at the same time?

Jason2866, 2025-10-19:
I think this is the commit which causes the seen behaviour https://github.com/arendst/Tasmota/pull/23553

sfromis, 2025-11-27:
My working device has `HDW: ESP32-D0WDQ6 v1.0`, same as your previous boot looping case. Your last case looks like the same, just with details of the boot loop. IDK what happened, but I'd not expect it to be about chip type.

I do notice the message `npl funcs init failed` just before the panic; it seems plausible that there is a link, but I don't even have a guess about the meaning of that message.

sfromis, 2025-12-03:
While my BLE device does have a PSRAM chip, it is not enabled due to the ESP32 being so old that Tasmota cannot use it without a build including a fix for this early revision hardware. Hence, no PSRAM, effectively, as I just used the standard precompiled build; same one as you were using.